serialport/node-serialport #1430: Can't install serialport -- EmergingTechnologyAdvisors doesn't list the package ??

otisbison, 2017-12-28:
### Summary of Problem

Can't install node-serialport because EmergingTechnologyAdvisors no longer provides the package!

### Steps and Code to Reproduce the Issue

Try installing -- and get a 404 when it tries to grab serialport. Go to EmergingTechnologyAdvisors on github and note that the package is no longer there.

reconbot, 2017-12-28:
Can you fill out the form about which version your using and what os? The downloads redirect from the old repo.

otisbison, 2017-12-28:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__:  9.3.0
- __Operating System__ and __Hardware Platform__: Windows 10 Pro 64 bit pc

I do: npm install serialport and it shows an error when it tries to fetch the Emerging[...] url. It’s as if the redirect isn’t occurring.

fivdi, 2017-12-28:
Please post the complete console output of the command `npm install serialport`.

fivdi, 2017-12-28:
One additional point, the command `npm install serialport` will install the latest version of serialport. Currently the latest version of serialport is v6.0.4. If for some reason you need to install serialport v4.0.7 the command to run is `npm install serialport@4.0.7`.

davidperrenoud, 2017-12-28:
It looks like the maximum supported version is Node 8 (v57), as the packages for Node 9 (v59) are not yet available:

```
$ npm install --save-exact serialport

> serialport@6.0.4 install /Users/david/Desktop/TestSerialport/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.4.1
prebuild-install info looking for local prebuild @ prebuilds/serialport-v6.0.4-node-v59-darwin-x64.tar.gz
prebuild-install info looking for cached prebuild @ /Users/david/.npm/_prebuilds/https-github.com-EmergingTechnologyAdvisors-node-serialport-releases-download-v6.0.4-serialport-v6.0.4-node-v59-darwin-x64.tar.gz
prebuild-install http request GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v59-darwin-x64.tar.gz
prebuild-install http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v59-darwin-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=9.3.0 runtime=node arch=x64 platform=darwin)
[...]
```

reconbot, 2017-12-28:
That's true, but it should build them for you if your computer is setup for
it.

---
Francis Gulotta
wizard@roborooter.com

otisbison, 2017-12-29:
That explains why it couldn't find the binary. 
It wasn't building the source either and I managed to figure out it was because of CL.EXE, which I now see is listed in your special Windows cases.

I don't think I was always getting that CL.EXE error, but maybe I just missed it. Anyway, sorry for the trouble. Getting those dev components installed through VS fixed it all up.

closed 2017-12-29 by otisbison

davidperrenoud, 2017-12-29:
@reconbot I think this should be reopened as it isn't compiled from source automatically, at least not in my case.

reconbot, 2017-12-30:
@davidperrenoud can you open your own issues with more details and the full output from npm? It should be building from source automatically.