serialport/node-serialport #1431: After random duration stops reading data until port reopened

akashbudhia, 2018-01-02:
- __SerialPort Version__: 
- __NodeJS Version__: v8.2.0
- __Operating System__ and __Hardware Platform__: Raspberry Pi

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No issues
- Are you using __Electron__ and have you checked the _Electron Docs_?: Not using Electron

### Summary of Problem
It happens all the time that after a certain duration the serial port stops reading any data at all until port is closed and opened again. And whenever this happens, it happens for all the serial ports at the same time. The ones which are reopened will start reading the data from the serial port while others will stay not responding.

### Please advise if any of you have experienced such behaviour before and some hints on when this would happen.

reconbot, 2018-01-02:
What version of SerialPort?

akashbudhia, 2018-01-02:
serialport@4.0.7

reconbot, 2018-01-02:
It's fixed in 6x

akashbudhia, 2018-01-02:
Do you have a any thread/commit which talks about this? With minimal code I'm not able to reproduce this issue.

I'm trying to rule out the possibility of any issue in my application code.

reconbot, 2018-01-02:
We fixed a few global state issues, and a few places where errors are not
cleaned up. I've never seen this exact issue (my apologies for not being
clear before) but these kinds of issues most likely live in rewritten
sections of the library.

akashbudhia, 2018-01-02:
I'm working on upgrading to 6x. Can you give an example of 'readable' with 'parsers' instead of [example](https://github.com/node-serialport/node-serialport/blob/master/examples/readline.js) with data event.

Is this correct?
```
this.device = new SerialPort(ports.path);
this.parser = this.device.pipe(new Readline({ delimiter: '\n' }));
//readable event on parser?
this.parser.on('readable', onData.bind(this));
```
and
```
function onData() {
    //.read on the device/serialport object
    var data = this.device.read();
    console.log(data);
}
```

reconbot, 2018-01-15:
You can use the `data` event even with serialport 6. But with your example you should keep calling read until it returns null.

lkishfy, 2018-08-23:
@reconbot I am currently having this exact same issue in v6.1.1. After about 6 hours, serialport stops reading data until port is closed and re-opened

closed 2018-08-29 by reconbot

NodeX-EG, 2019-01-11:
I have exactly the same problem. Everything seems to be fine, but at certain point (after like day or two) it stops reading serial input.

lkishfy, 2019-01-11:
Hi @NodeX-EG I was never able to resolve what both you and I observed. For the project I was working on I ended up emitting the data from serial to a Node app via powershell instead of using this module. 

The project I was working on has been installed for months now, running 8 hours a day and I haven't run into any issues. Hope this helps you with your project.

NodeX-EG, 2019-01-11:
I am running all in Raspberry pi... No PS here. I will see what I can do about it. But this messes up my whole project. I was thinking it might be Arduino actually stops relaying messages, but after testing Arduino without NodeRed Serial for months - it was working just fine.