serialport/node-serialport #1433: installation of serialport fails during build of serialport.cpp

mauermbq, 2018-01-07:
- __SerialPort Version__:serialport@6.0.4 i
- __NodeJS Version__: 5.6.0
- __Operating System__ and __Hardware Platform__: rasperian 4.9.35-v7+

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem
installation of serialport fails during build of serialport.cpp (see below)
### Steps and Code to Reproduce the Issue
call sudo npm install serialport --unsafe-perm --build-from-source
output:
> serialport@6.0.4 install /home/pi/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.4.1
prebuild-install info install --build-from-source specified, not attempting download.
make: Entering directory '/home/pi/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:79:1: internal compiler error: Speicherzugriffsfehler
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
Preprocessed source stored into /tmp/ccq75Oj4.out file, please attach this to your bugreport.
serialport.target.mk:95: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/pi/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.9.35-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/serialport
gyp ERR! node -v v8.9.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.0.4 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1

fivdi, 2018-01-07:
If I'm not mistaken the error message `"internal compiler error: Speicherzugriffsfehler"` which is `"internal compiler error: Memory access error"` in English means that the compiler is having problems, not serialport.

Everything looks ok for me on a Raspberry Pi 3.

```
pi@raspberrypi:~/serialport $ uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
pi@raspberrypi:~/serialport $ node -v
v8.9.4
pi@raspberrypi:~/serialport $ npm -v
5.6.0
pi@raspberrypi:~/serialport $ npm install serialport
npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EAI_AGAIN: request to https://registry.npmjs.org/promirepl failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org:443
npm WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.

> serialport@6.0.4 install /home/pi/serialport/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.4.1
prebuild-install info looking for local prebuild @ prebuilds/serialport-v6.0.4-node-v57-linux-arm.tar.gz
prebuild-install info looking for cached prebuild @ /home/pi/.npm/_prebuilds/https-github.com-EmergingTechnologyAdvisors-node-serialport-releases-download-v6.0.4-serialport-v6.0.4-node-v57-linux-arm.tar.gz
prebuild-install http request GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v57-linux-arm.tar.gz
prebuild-install http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v57-linux-arm.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=8.9.4 runtime=node arch=arm platform=linux)
make: Entering directory '/home/pi/serialport/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/serialport/node_modules/serialport/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/serialport/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/serialport/package.json'
npm WARN serialport No description
npm WARN serialport No repository field.
npm WARN serialport No README data
npm WARN serialport No license field.

+ serialport@6.0.4
added 58 packages in 116.673s
pi@raspberrypi:~/serialport $ 
```

The command `sudo npm install serialport --unsafe-perm --build-from-source` was used for the install. I wonder if the issue is related to the fact that `sudo` is being used. It looks like you were logged on as user pi for the install. Is this the case? If so, why is `sudo` being used as it shouldn't be necessary?

Can you logon as user pi and try running the following commands to see if it works please:
```
pi@raspberrypi:~ $ mv node_modules node_modules_backup
pi@raspberrypi:~ $ mkdir serialport
pi@raspberrypi:~ $ cd serialport
pi@raspberrypi:~/serialport $ npm install serialport
```

The command `mv node_modules node_modules_backup` is called to prevent the modules that are already installed from getting in the way.

mauermbq, 2018-01-07:
> 
If I'm not mistaken the error message "internal compiler error: Speicherzugriffsfehler" which is "internal compiler error: Memory access error" in English means that the compiler is having problems, not serialport.
> 
yes, you are right. (sorry with regard to German spelling, just copied the output 1:1) 

compiler version: gcc 4.9.2 (Raspbian 4.9.2-10) 

However, `npm install serialport`  was my first try, and both ended up with the same compilation problem of `serialport.cpp`
 
Here the output following your command instructions:
> serialport@6.0.4 install /home/pi/serialport/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.4.1
prebuild-install info looking for local prebuild @ prebuilds/serialport-v6.0.4-node-v57-linux-arm.tar.gz
prebuild-install info looking for cached prebuild @ /home/pi/.npm/_prebuilds/https-github.com-EmergingTechnologyAdvisors-node-serialport-releases-download-v6.0.4-serialport-v6.0.4-node-v57-linux-arm.tar.gz
prebuild-install http request GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v57-linux-arm.tar.gz
prebuild-install http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v57-linux-arm.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=8.9.4 runtime=node arch=arm platform=linux)
make: Entering directory '/home/pi/serialport/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:79:1: internal compiler error: Speicherzugriffsfehler
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
Preprocessed source stored into /tmp/ccMKEBao.out file, please attach this to your bugreport.
serialport.target.mk:95: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/pi/serialport/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.9.35-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/serialport/node_modules/serialport
gyp ERR! node -v v8.9.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/serialport/package.json'
npm WARN serialport No description
npm WARN serialport No repository field.
npm WARN serialport No README data
npm WARN serialport No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.0.4 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.0.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-01-07T18_34_18_098Z-debug.log

fivdi, 2018-01-07:
The translation of "Speicherzugriffsfehler" to "Memory access error" is too literal a translation and the correct translation is more likely to be "Segmentation fault". It's gcc that's segfaulting and the error message also suggests submitting a gcc bug report.
```
../src/serialport.cpp:79:1: internal compiler error: Speicherzugriffsfehler
}
^
Please submit a full bug report,
with preprocessed source if appropriate.
See file:///usr/share/doc/gcc-4.9/README.Bugs for instructions.
Preprocessed source stored into /tmp/ccMKEBao.out file, please attach this to your bugreport.
```
I may be wrong but I would imagine that it's more likely that something is wrong with the Raspbian installation on your Pi than there being a bug in gcc.

Can you flash the latest version of Raspbian onto an SD card and try with that? The latest version of Raspbian comes with gcc 6.3.0 rather than 4.9.2.

Although it's not going to help you could also try installing a few other native addons to see if there is also a Speicherzugriffsfehler with them, for example:
```
npm install onoff i2c-bus spi-device
```

mauermbq, 2018-01-10:
upgrading to jessie solved the problem - thank you!!

fivdi, 2018-01-10:
> upgrading to jessie solved the problem

Excellent

> thank you!!

Your welcome :smile:

closed 2018-01-10 by fivdi