serialport/node-serialport #1436: can't install serialport on windows

mikeGHQ, 2018-01-13:
- __NodeJS Version__: 9.4.0
- __Operating System__ and __Hardware Platform__: windows x64

### Summary of Problem
i can install serialport on linux without no problem, but on windows
i tried `npm install serialport` but i get this output

```
> serialport@6.0.4 install C:\Users\Mohammed\node_modules\serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.4.1
prebuild-install info looking for local prebuild @ prebuilds\serialport-v6.0.4-node-v59-win32-x64.tar.gz
prebuild-install info looking for cached prebuild @ C:\Users\Mohammed\AppData\Roaming\npm-cache\_prebuilds\https-github.com-EmergingTechnologyAdvisors-node-serialport-releases-download-v6.0.4-serialport-v6.0.4-node-v59-win32-x64.tar.gz
prebuild-install http request GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v59-win32-x64.tar.gz
prebuild-install http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v59-win32-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=9.4.0 runtime=node arch=x64 platform=win32)

C:\Users\Mohammed\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
MSBUILD : error MSB4132: The tools version "2.0" is unrecognized. Available tools versions are "12.0", "4.0".
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\12.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:160:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:12)
gyp ERR! System Windows_NT 10.0.16299
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Mohammed\node_modules\serialport
gyp ERR! node -v v9.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\Mohammed\package.json'
npm WARN Mohammed No description
npm WARN Mohammed No repository field.
npm WARN Mohammed No README data
npm WARN Mohammed No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.0.4 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.0.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Mohammed\AppData\Roaming\npm-cache\_logs\2018-01-13T12_14_17_593Z-debug.log

```

mikeGHQ, 2018-01-13:
i tried to install in my project folder (next to package.json) but i get this error

```
> serialport@6.0.4 install C:\Users\Mohammed\Desktop\Ard.Rem.App\ardiono-remote-app\node_modules\serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.4.1
prebuild-install info looking for local prebuild @ prebuilds\serialport-v6.0.4-node-v59-win32-x64.tar.gz
prebuild-install info looking for cached prebuild @ C:\Users\Mohammed\AppData\Roaming\npm-cache\_prebuilds\https-github.com-EmergingTechnologyAdvisors-node-serialport-releases-download-v6.0.4-serialport-v6.0.4-node-v59-win32-x64.tar.gz
prebuild-install http request GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v59-win32-x64.tar.gz
prebuild-install http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v59-win32-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=9.4.0 runtime=node arch=x64 platform=win32)

C:\Users\Mohammed\Desktop\Ard.Rem.App\ardiono-remote-app\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
MSBUILD : error MSB4132: The tools version "2.0" is unrecognized. Available tools versions are "12.0", "4.0".
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\12.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:160:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:12)
gyp ERR! System Windows_NT 10.0.16299
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Mohammed\Desktop\Ard.Rem.App\ardiono-remote-app\node_modules\serialport
gyp ERR! node -v v9.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN arduino-remote@ No repository field.
npm WARN arduino-remote@ No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.0.4 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.0.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Mohammed\AppData\Roaming\npm-cache\_logs\2018-01-13T12_36_33_216Z-debug.log

C:\Users\Mohammed\Desktop\Ard.Rem.App\ardiono-remote-app>cd ardiono-remote-app
```

closed 2018-01-13 by mikeGHQ

hanfengcan, 2018-01-15:
`http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.4/serialport-v6.0.4-node-v59-win32-x64.tar.gz`

try the cnpm

or

download serialport-v6.0.4-node-v59-win32-x64 from https://github.com/node-serialport/node-serialport/releases and save in mpn _prebuilds folder

reconbot, 2018-01-15:
We don't have builds for node 9, we do for node 8 however

---
Francis Gulotta
wizard@roborooter.com