serialport/node-serialport #1437: how to check if arduino drivers are installed

mikeGHQ, 2018-01-13:
this is not a problem in node-serialport, but i want to check if i can do this using serialport?

i created this  code that will search for all serialports and detect if there is an arduino connected the open its port automatically as below:

```
var SerialPort = require('../node_modules/serialport');
var arduinoport;
function getConnectedArduino() {
  SerialPort.list(function (err, ports) {
    ports.forEach(function(port) {
      pm = port['manufacturer'];
      if (typeof pm !== 'undefined' && pm.includes('arduino')){
        arduinoport = port.comName.toString();
        var serialPort = require('../node_modules/serialport');
        sp = new serialPort(arduinoport, {buadRate: 9600})
        sp.on('open', function () {
          portready(arduinoport)
        })
        return;
      }
    });

  });
}
``` 

now this code work if i have arduino drivers installed - of course - but how to show the user a message that he have to install arduino drivers if the module couldn't find and arduino port??

i think i can do this using pure node.js but i am new to it :D so can someone please help me :D

reconbot, 2018-01-15:
I don't know how to detect if the drivers are installed (windows I'm guessing?) and it's not something that nodejs can do easily.

Because you have serialport in your node_modules, you can require it by using
```js
const SerialPort = require('serialport')
````
And that will work in any directory within your project.

mikeGHQ, 2018-01-16:
thank you @reconbot for  the help i didn't know that i can require a module like this :D, by the way i solved this problem using this code. it will count the ports then if the `forEach` statement was executed by the ports count number and it ddn't found arduino. it will alert you that you have to install the drivers or check if arduino is connected
<hr>
```
function getConnectedArduino() {
  $('#searchProcess').empty()
  SerialPort.list(function(err, ports) {
    var allports = ports.length;
    var count = 0;
    var done = false
    ports.forEach(function(port) {
      count += 1;
      pm = port['manufacturer'];
      if (typeof pm !== 'undefined' && pm.includes('arduino')) {
        arduinoport = port.comName.toString();
        var serialPort = require('../node_modules/serialport');
        sp = new serialPort(arduinoport, {
          buadRate: 9600
        })
        sp.on('open', function() {
          portready(arduinoport)
        })
        sp.on('close', function (err) {
          alert('i think you disconnected the arduino!');
          window.location.reload()
        })
        done = true;
      }
      if (count === allports && done === false) {
        alert('please make sure that arduino is connected and you installed the drivers ')
      }
    });

  });
}
``` 

so i hope this will help somebody :D

closed 2018-01-16 by mikeGHQ

mikeGHQ, 2018-01-16:
and if any body is interested here is a link to the nwjs app https://github.com/mikeGHQ/arduino-remote-linux