serialport/node-serialport #1438: Not receiving data properly. Getting 16 bit data instead of 8 bit.

leitscoper, 2018-01-13:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: v8.8.1
- __Operating System__ and __Hardware Platform__: Windows 7 64 bit, x64 bit CPU, Laptop

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: Yes

### Summary of Problem
I am sending following data with my arduino:
[https://i.imgur.com/ZmPHyPC.png](https://i.imgur.com/ZmPHyPC.png)

But I receive:
[https://i.imgur.com/RcAbC8o.png](https://i.imgur.com/RcAbC8o.png)
Those "fffd" are weird, they shouldn't be there.

### Steps and Code to Reproduce the Issue

Posted above.
`console.log(stringToHex(zmienna_duza));`
zmienna_duza is raw data received by serial port. It is a string.

fivdi, 2018-01-15:
@leitscoper The code posted above doesn't contain any serialport code. Are you 100% sure that `zmienna_duza` is a string and that it is not a Node.js [Buffer](https://nodejs.org/dist/latest-v8.x/docs/api/buffer.html#buffer_buffer)? I'd expect `zmienna_duza` to be a Buffer which is basically an array of bytes. If `zmienna_duza` is a Buffer then the problem is probably in `stringToHex`.

leitscoper, 2018-01-15:
I can give you the code.
`var zmienna_duza = "";` declaration of variable, on top of renderer.js

`var serialport = require('serialport')
const createTable = require('data-table')

var all_ports = ["1", "2", "3", "4"]; // nwm jak sie robi puste arraye więc tak bywa
var selected_port = ""; // aktualnie wybrany port, jeżeli wynosi 68, nie jest wybrany
var SerialPort = require("serialport");

var raw_data = "";
var data_converted = "";
var temporary_data = "";
var zmienna_duza = "";` More detailed

fivdi, 2018-01-15:
This only means that `zmienna_duza` starts off as a string. By the time `stringToHex` is called it could be a Buffer.

leitscoper, 2018-01-15:
`tablica = new Uint8Array(80);
	for(var x = 0; x < 80; x++)
	{
		tablica[x] = dane_do_analizy.charCodeAt(x);
		console.log ( (tablica[x]).toString(16) );
	}`
that prints it.

fivdi, 2018-01-15:
I'm afraid I don't understand. Can you post a complete program using serialport that can be used to reproduce the problem please? The shorter that program is the better.

leitscoper, 2018-01-16:
Yes, I can give you while program. Should be around 18:00 GMT+1

fivdi, 2018-01-16:
@leitscoper please keep the program that can be used to reproduce the issue as short as possible.

fivdi, 2018-01-20:
@leitscoper provided sample code and it turned out that this issue isn't related to serialport.

closed 2018-01-20 by fivdi