serialport/node-serialport #1439: Get Serialport 6 work (again) on ARM systems with Linux

Apollon77, 2018-01-14:
I tried to use Serialport 6.x in my projects but have the same problems then others. It just don't work. you do not get any data when reading.

My System is a Cubietruck with an Armbian (Debian Stretch with Mainline Kernel), but also raspberries had such a problem as I get feedback.

What is possible to to to get back Support for these systems for Serialport 6 (4 works ok-ish ... but has many issues as you know)?!
How can I support?

fivdi, 2018-01-14:
@Apollon77 That sound frustrating. The issue template contains the fields and sections listed below that need to be filled out. If the requested information is provided it will increase the chances of someone being able to provide assistance.

- __SerialPort Version__:
- __NodeJS Version__:
- __Operating System__ and __Hardware Platform__:

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem

### Steps and Code to Reproduce the Issue

Apollon77, 2018-01-15:
I updated the issue as requested

fivdi, 2018-01-15:
@Apollon77 Please take a look at the latest comments in #1415 to see if it helps. If I'm not mistaken this issues is a duplicate of #1415 and can be closed, correct?

Apollon77, 2018-01-15:
I will test that later this afternoon and report. 

I decided to open a new issue because Ibise an ARM device with Linux and the other issue was not talking about arm ... and arm was listed as „should be but untested“. 

After my test we know it ;-)

reconbot, 2018-01-15:
https://github.com/node-serialport/node-serialport/blob/master/README.md#platform-support
basically says that 😋

reconbot, 2018-01-15:
Ohh I undestand! Arm works well but we don't have any CI tests for ARM and we don't provide any builds for ARM because of nodejs lack of support for detecting what version of ARM the system is and travis ci's lack of support for ARM systems. There might be a way to test with docker containers but it is not yet proven.

Apollon77, 2018-01-15:
So, 2s waiting will not make any difference (as thought). When I go to serialport 4.0.7 it works, it i use the exact same code with serialport 6.0.4 it get no response

Apollon77, 2018-01-15:
If it helps I can do  second run with 6.0.4 and try to enable Serialport debug (Hope I get that done, or I need to find a short example without my full code ...)?

Apollon77, 2018-01-15:
I found my problem :-( Was mine fault.

I had an "update" statement to set a baudrate and here is forgot to use the "new" case "baudRate" ... and so it was set to the wrong baudrate :-(

closed 2018-01-15 by Apollon77

fivdi, 2018-01-15:
@Apollon77 good to hear that it's working now and thanks for taking the time to provide feedback.