serialport/node-serialport #1440: install particle-cli fail..

kdupreez, 2018-01-14:
serialport@6.0.4
node -v v9.4.0
Ubuntu 16.04 LTS (x64)

### Summary of Problem
Trying to install Particle CLI through "npm install -g particle-cli" and keep getting this error:

kdupreez@imacbuntu:~/software$ sudo npm install -g particle-cli
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
/usr/bin/particle -> /usr/lib/node_modules/particle-cli/bin/particle.js

> serialport@6.0.4 install /usr/lib/node_modules/particle-cli/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.4.1
prebuild-install info looking for local prebuild @ prebuilds/serialport-v6.0.4-node-v59-linux-x64.tar.gz
prebuild-install WARN install EACCES: permission denied, access '/home/kdupreez/.npm'
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/particle-cli/node_modules/serialport/build'
gyp ERR! System Linux 4.13.0-26-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/particle-cli/node_modules/serialport
gyp ERR! node -v v9.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.0.4 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
**npm ERR! Failed at the serialport@6.0.4 install script.**
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/kdupreez/.npm/_logs/2018-01-14T22_11_26_828Z-debug.log

reconbot, 2018-01-14:
Should work with node 8

fivdi, 2018-01-14:
Or perhaps it's because the option `--unsafe-perm` wasn't specified.
See [sudo / root section](https://github.com/node-serialport/node-serialport#sudo--root) section of readme.

kdupreez, 2018-01-15:
fivdi - I can install serialport by itself just fine. but the particle-cli package seems to have its own serialport package and that fails..  
 
So I didnt think of adding the --unsafe-perm option onto the particle-cli install..  I did that and BOOM! it worked! 

THANK YOU!

closed 2018-01-15 by reconbot