serialport/node-serialport #1450: How to reset a serialport connection

sgupta999, 2018-01-23:
- __SerialPort Version__:6.0.4
- __NodeJS Version__:4.8.2
- __Operating System__ and __Hardware Platform__: Original Raspberry Pi 1 (armv6)

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
I am using serial port to connect to a Rainforest EMU2 energy monitoring device connecting to the USB port on the raspberry pi.
The connection works fine and I am able to connect to device and receive data.
However the device stops sending data after 15 minutes.
If I kill my process and restart the process the device will connect again and start broadcasting.
instead of trying to kill the process and restart I am trying to just close the serialport connection. start a new connection. However that does not seem to be working. How can I simulate the same effect of connecting to the device with a new serialport as if I am restarting my process.

I have tried deleting seriaport module from cache and reloading but nothing seems to be working.
Although the serialport shows open and the open method is called no data is being received from the device. While if i kill process and restart immediately the device begins transmitting the data.

### Steps and Code to Reproduce the Issue

	console.log("Resetting port at ..." +  new Date());
	if (self.serialPort.isOpen) {
		//console.log("Port is open ...");
		self.serialPort.close();
	}	

	serialport = requireUncached("serialport"); 
	portparser  = new serialport.parsers.Readline;
	self.serialPort = new serialport(state.serialPath, {
        baudRate: 115200,
        dataBits: 8,
        stopBits: 1,
        parity: 'none',
    });
	self.serialPort.pipe(portparser);
	self.serialPort.on("open", function() {	
		openHandler(self);
    })
	
	//Using this uncache fuction to force reload of serialport
	
	function requireUncached(module){
    delete require.cache[require.resolve(module)]
    return require(module)
	}

reconbot, 2018-01-24:
This sounds a bit like a memory or file handle leak. It should be noted
that Close is an async operation, you might try waiting for the port to
close before opening it in another instance.

Are you creating a new port each time or reusing this one?

sgupta999, 2018-01-24:
Creating a new port each time.
I put all close commands in a separate function and calling openport with a 30 seconds timeout.
so 30 seconds between close and open (I have tried to change it to 60 no difference reallly). if i Ctrl C and kill process and restart (all within 10 seconds everything is working. Same behaviour on Rpi 1, Windows 10 and Rpi 3 - in RPi 3 I am actually able to reset.

Clearing serialport from cache so assuming no memory leaks but i could be wrong

function requireUncached(module){
    delete require.cache[require.resolve(module)]
    return require(module)
}

function resetPort(){	
	state.PORT_INITIALIZED = false;
	if (self.serialPort.isOpen) {
		winston.debug("Port is open ...closing");
		self.serialPort.close();
	}	
	setTimeout(openPort, 30000);
}

function openPort(){
	try {
		winston.info("Opening port");
	        serialport = requireUncached("serialport");
		portparser  = new serialport.parsers.Readline;
		self.serialPort = new serialport(state.serialPath, {
			baudRate: 115200,
			dataBits: 8,
			stopBits: 1,
			parity: 'none',
		});
		self.serialPort.pipe(portparser);
		self.serialPort.on("open", function() {	
			openHandler(self);
		});
	} catch (ex){
		winston.error("ERROR opening port\n" + ex);
	}
}

sgupta999, 2018-01-24:
OK udpate - i found one issue 
i had defined portparser as const in the top as originally i was not expecting to have to reset port and parsers. Fixing that now allows me to reset the port of Rpi 1 also. 

However the port stops receiving data on all 3 installation between 15-19 minutes.. I need to figure out how to fix this  - any help will be appreciated

reconbot, 2018-02-05:
I've never had to mess with the require cache. If there's any state being kept it will be in the c++ module and there's no way to clear that.

It's not clear to me when `resetPort` is called.

You wrote `Failed Poll: resetting port` in the logs, how does that happen?

sgupta999, 2018-02-05:
the device broadcasts an event every 15 seconds which triggers the call to the port 'data' callback function
In the ports 'data' callback function - as soon as you enter the function (i.e. the port received an event from the device) a flag is set signifying connection is working and data is being received.

I have another thread that polls that flag every 15 seconds if - if the flag is not set it increments a counter and after 5 failed tries it calls the reset port method.

my workaround was working but after 3 days the port again became unresponsive. It wouldn't respond to even a restart of the program. I had to physically remove the USB connection and reconnect it again and it has been working since.

Very frustrating and not sure how to diagnose where the issue is.

reconbot, 2018-09-20:
I'm not convinced it's not a USB driver issue. The read code between windows and osx are not significantly different. We use nodejs's fs.read under the hood which is super well tested.

I wish I had a better answer here.

sgupta999, 2018-09-20:
Fact is the workaround seems to be working for now - I reset the connection everytime I detect that data is not being parsed. Sometime even the reset will not work for a few cycles but eventually it is able to reset and gets the data.

still very odd  - definitely looks like a problem somewhere with the driver or the connection

closed 2018-11-27 by stale[bot]

krzychu0905gmail, 2018-12-22:
i have the same problem with port after receiving more than ~32768 bytes no more 'on data' event. Closing and reopening port not resolving problem. I must restart application.

reconbot, 2018-12-22:
Is that a consistent number? What version, platform, os, etc

reconbot, 2018-12-23:
Please open a new issue, and try with the latest version of serialport