serialport/node-serialport #1455: Best practices on how to handle external signals ('SIGINT', 'SIGUSR1', 'SIGUSR2','SIGTERM') and avoid locked ports (graceful app closing)

mreis1, 2018-01-26:
I'm running an app on osx. I would like to close every open serialport before the process exits otherwise I'll find myself with constant locked ports. Is there a recommended way to handle this?

reconbot, 2018-01-29:
If node exits it should close all ports.

mreis1, 2018-01-30:
That's weird @reconbot because opening a port with the "lock" option set to true and closing the process by sending a SIGINT (Ctrl+C) will lock the port to future node processes.
I'm using node-serialport version "5.0.0".

daganchen, 2018-02-04:
I have also used this option, but it doesn't work on windows!

reconbot, 2018-02-05:
@mreis1 you might want to put a `port.close()` in the sigint handler and see if that changes anything. `port.close()` doesn't do any unlocking however. This isn't the kernel manual but we use 'flock' and [https://stackoverflow.com/a/10987644/339](unlocking isn't usually needed). Maybe the system needs more time?

@daganchen you can use a callback to handle errors on open. Can you explain more?

mreis1, 2018-02-09:
@reconbot I noticed a weird behaviour when **attempting to open a serialport that somehow is 'locked'.**
If I try to call 'open' method the 'open' event or even the `open` callback will never get fired and will as consequence lock the node process. During this state calling port.close() will throw an error saying that the port is not (yet) open which is correct, but it would be great to have a way to interrupt the ongoing connection attempt, for example, through a timeout?

mreis1, 2018-02-09:
I would like to point out that moving the dev environment to a windows virtual machine seems to have fixed the issue, at least I know that the OSX module is not stable enough when compared to his sibling windows version.

During development I find myself connecting to 3 different serialport devices, since I can't predict which device is using a specific port before the application is started, I've to perform a scan that consists in sending a given command and expect the device to understand the response, and if does I will map it to a particular device inside my application.

This scan process, needs to open and close the same port many times and while windows seems to handle it correctly, my mac in the other hand will eventually crash and lock my serialport or even the USB completely (not even a USB keyboard will work in there). When this happens I'm forced to restart the system.

reconbot, 2018-02-28:
I've seen this crash from opening and closing from buggy device drivers. It really depends on which adaptors you have and how well they're supported on OSX.

closed 2018-11-17 by reconbot