serialport/node-serialport #1459: Wrong serial number on CDC ACM composite devices on win32

IvanSanchez, 2018-01-30:
This is kind of a follow-up of #1220.

I'm using `serialport` v6.0.4 (latest available as of 2018-01-28) on nodejs 8.9. I have a Linux (debian sid) box, and a win32 (Windows 10 v1607 build 14393.2007) box.

The serial port device I'm working with is a [nRF52840 preview development kit](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52/dita/nrf52/development/nrf52840_pdk/kit_hw_content.html?cp=2_0_3_2_0), loaded with firmware [implementing USB CDC ACM](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v14.2.0/lib_usbd_class_cdc.html?cp=4_0_0_3_55_8_3) for serialport-like communications. It's basically a beefed-up arduino-like board with wireless stuff and its own fully-configurable USB stack.

### Summary of Problem

In my case, the serial number of the USB device is set up when compiling the firmware, and is used by the USB stack to populate the `iSerial` field of the USB descriptor. This is how the serial number looks over the wire (wireshark screenshot):

[image]

And this is how a `lsusb -v` looks like in my linux box (relevant part):

```
Device Descriptor:                                                                                                                                                                                                                                             
  idVendor           0x1915 Nordic Semiconductor ASA                                                                                                                                                                                                                          
  idProduct          0x521f                                                                                                                                                                                                                                                   
  bcdDevice            1.00                                                                                                                                                                                                                                                   
  iManufacturer           1 Nordic Semiconductor                                                                                                                                                                                                                              
  iProduct                2 nRF52 USB CDC Demo                                                                                                                                                                                                                                
  iSerial                 3 bf8f64cd7b5be46c             
```

And this is how `node-serialport/bin/list.js -f jsonline` looks like in linux (with extra line breaks for reading pleasure):

```json
{
"manufacturer":"Nordic Semiconductor",
"serialNumber":"bf8f64cd7b5be46c",
"pnpId":"usb-Nordic_Semiconductor_nRF52_USB_CDC_Demo_bf8f64cd7b5be46c-if00",
"vendorId":"1915",
"productId":"521f",
"comName":"/dev/ttyACM0"
}
```

So in Linux, I get to see the serial number of the USB device as the serial number. This is good. This is expected.

Now I switch to win32.

The output of  `node-serialport/bin/list.js -f jsonline` in win32 looks like:

```json
{
"comName":"COM5",
"manufacturer":"Microsoft",
"serialNumber":"7&307B223F&1&0000",
"pnpId":"USB\\VID_1915&PID_521F&MI_00\\7&307B223F&1&0000",
"locationId":"0000.0014.0000.005.002.000.000.000.000",
"vendorId":"1915",
"productId":"521F"}
```

Win32 reports the serial number as `7&307B223F&1&0000`, but it should be `bf8f64cd7b5be46c`... so where does that come from?

The answer is in how win32 handles composite USB devices. If I open up win's device manager, it looks like:

[image]

If I fire up `regedit.exe` to see details of those two devices (under `HKEY_LOCAL_MACHINE/System/CurrentControlSet/Enum/USB`, I can see...

[image]

Well, that makes sense. One of the devices maps to the USB device as a whole, and the other maps to the CDC ACM interface. And the naming makes sense according to https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/enumeration-of-interfaces-not-grouped-in-collections .

Digging a bit deeper, I can see how the drivers are bound to "devices":

[image]

So my guess is that `serialport` is interacting with the `usbser.sys` kernel driver, fetching information from it - but the right serial number is not available at that level.

The question is: **How can `serialport` check if a serial port is part of a USB composite device, and if so how can it fetch the parent's serial number?**

### Steps and Code to Reproduce the Issue

Get a hold of a USB composite device with a CDC ACM interface, play around with it with the windows device manager as shown above :-)

---

Addendum: It is possible to force win32 to bind the `usbser.sys` to the USB device as a whole: run Zadig, show all devices, choose the composite device, replace `usbccgp.sys` with `usbser.sys` (AKA `USB Serial (CDC)`)". In this case, things will look like:

[image]

...and `serialport` will be able to fetch the right serial number. However, this prevents other interfaces in the device from working (e.g. a device with CDC ACM and mass storage at the same time), and requires manual user intervention with scary messages, so it's just a stopgap measure.

reconbot, 2018-02-03:
This is great research, I learned something about how windows handles these devices.

> The question is: How can serialport check if a serial port is part of a USB composite device, and if so how can it fetch the parent's serial number?

This is the question. The code responsible [can be found here](https://github.com/node-serialport/node-serialport/blob/master/src/serialport_win.cpp#L600-L695) and we go a step further parsing it a little more in [JS here](https://github.com/node-serialport/node-serialport/blob/master/lib/bindings/win32-sn-parser.js) to have a consistent output. Even if you're not familiar with c++ windows keeps consistent names between their api calls in different languages. So you can google the function names (eg [SetupDiGetClassDevs](https://msdn.microsoft.com/en-us/library/windows/hardware/ff551069(v=vs.85).aspx)) to get more info.

Figuring out the answer to your question is the first step to improving this feature. =)

IvanSanchez, 2018-02-05:
One of my colleagues has been playing around with making more lookups to the win32 registry; it should be possible to iterate through the USB device tree looking for matching serial numbers, and see if the "device" is a USB interface or the USB device as a whole.

*If* our research is successful, we *should* have a PR ready in like a week or so. 🤞

IvanSanchez, 2018-02-15:
Progress! Some of our code can already iterate through the registry keys, and match the [container ID]() of the device nodes for the ports.

[image]

The code is ugly, the algorithm is clunky, I have no f***ing clue how to compare a `char*` to a `WCHAR*`, but hey, it's looking good.

PR #1483 (2018-02-28): Query registry for USB serial numbers in win
Fixes #1459.

This PR makes the code for win32 fetch the [ContainerID](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/container-ids) of the device node holding the serial port, then make a registry lookup for the composite USB device with a matching ContainerID. The serial number from the device node of the composite device is then returned and used.

The code might not be the cleanest ever, but it's in a working state.

Special kudos to @CoqRogue for all the work put on this.
closed 2018-02-28 by reconbot