serialport/node-serialport #1465: Error With Disconnecting Ports

flakjacket95, 2018-02-05:
- __SerialPort Version__: 6.0.5
- __NodeJS Version__: 8.9.3
- __Operating System__ and __Hardware Platform__: MacOS High Sierra

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Not an installing issue
- Are you using __Electron__ and have you checked the _Electron Docs_?: Yes

### Summary of Problem
Connected serialports work fine in general when actually connected but, disconnection is not working properly.
### Steps and Code to Reproduce the Issue
For me, using a basic electron app - connect to a serial port, communicate a bit (read and write a few bytes), and then unplug the USB cable connecting the device to the computer. Serialport is not detecting this, even though the associated tty file in /dev/ is removed. 
I'm tried to detect disconnects using the `error`, `disconnect`, and `disconnected` events. I also attempt to read and write to the serialport and then check the `isOpen` value.

With the USB cable to the serial device unplugged, I receive no calls to the events above but, the `error` event is properly called if I provide improper arguments to a function, for example `sp.write(1)` so, the error handler is functioning properly.

A call to the `write()` function returns `true`.
A call to the `read()` function returns `undefined`, regardless of the length of bytes requested.
The `isOpen` property never changes from `true`.

A few months ago, I did not have this issue and disconnection worked properly, I have not attempted to determine where the change occurred yet.

reconbot, 2018-02-05:
There is no longer a disconnect event, but the close event should fire on disconnect with an error object. Could you provide sample code?

flakjacket95, 2018-02-08:
I updated to the `close` event and I have the same issue. As an important note however, on Windows this does not cause an issue. The code below functions properly on Windows but not on Mac. 
```
var sp;
var serialPort = require("serialport");
var sp = new serialPort(name,{
   baudRate: 115200
});
sp.on("error", function () {
  console.log("Error.");
});
sp.on("close", function () {
  console.log("Disconnected.");
});
```

Like I mentioned before, the `close` event is never fired in this case.

reconbot, 2018-02-08:
In order for removal to be detected you need to be reading or writing

---
Francis Gulotta
wizard@roborooter.com

flakjacket95, 2018-02-09:
My apologies, I removed that code from my post above to keep it smaller. Didn't think it'd be necessary. I can provide some example code if necessary but, my application does read and write to the serial port after connection. The `close` event is still not fired.

ccurtis0, 2018-09-11:
I'm having the same problem (I think) under Ubuntu 18.04.
Node version 8.10.0
Serialport version 7.0.2

My serial port is a USB device plugged into a USB hub.
I execute:
    ./node_modules/serialport/bin/terminal.js -p /dev/ttyUSB0

Using an external loopback, I see the characters I type.
I then power off the hub.
Entries in /dev/ disappear but 'terminal.js' reports no error.
I can continue to type but no characters are received.
^C exits.

I didn't dig into the source for 'terminal.js', but my code likewise receives neither 'error' nor 'close' events when the USB device is disconnected.

It looks like this used to be detected, judging by this bug:
    https://github.com/node-serialport/node-serialport/issues/1243

I'm wondering if the fix went too far and is masking the error condition.

ccurtis0, 2018-09-12:
Please ignore my comment above. After investigating further I found:
a) terminal.js does not listen for the 'close' event
b) My code was not listening for 'data' events, as you mentioned was necessary.

I first noticed this when using version 4, so I upgraded to version 7. But before the upgrade I added some test code to force an error state, and that code didn't try to read. Version 7 on Linux appears to work as expected.

reconbot, 2018-09-12:
Awesome, if you could please file a bug for the terminal cli

closed 2018-11-18 by stale[bot]