serialport/node-serialport #1469: if node-serial port support multithread ？

ericdai, 2018-02-07:
- __SerialPort Version__:6
- __NodeJS Version__:8.9.4
- __Operating System__ and __Hardware Platform__:
windows and arm linux
- 
### Summary of Problem

### Steps and Code to Reproduce the Issue

reconbot, 2018-02-07:
Node doesn't really support multithread. Can you be more specific?

rafamedina, 2018-02-23:
Maybe he is talking about a web worker

reconbot, 2018-02-23:
Does node have a webworker yet?

rafamedina, 2018-02-23:
You are right, but talking about a child process, have you ever test it using them?

And other question @reconbot 
Is possible to change the com port dynamically?
In my particular case, I have many devices connected and the system has to select the right port depending on the business logic.

reconbot, 2018-02-23:
Change, no, but you can always make a new serialport object.

---
Francis Gulotta
wizard@roborooter.com

closed 2018-02-28 by reconbot