serialport/node-serialport #1470: Compile error

piechade, 2018-02-07:
- __SerialPort Version__: 6.1.0
- __NodeJS Version__: 9.5.0
- __Operating System__ and __Hardware Platform__: Alpine Linux Docker

### Summary of Problem

  CXX(target) Release/obj.target/serialport/src/serialport_linux.o
../src/serialport_linux.cpp: In function 'int linuxSetCustomBaudRate(int, unsigned int)':
../src/serialport_linux.cpp:10:18: error: 'TCGETS2' was not declared in this scope
     if(ioctl(fd, TCGETS2, &t)) {
                  ^~~~~~~
../src/serialport_linux.cpp:18:18: error: 'TCSETS2' was not declared in this scope
     if(ioctl(fd, TCSETS2, &t)) {
                  ^~~~~~~
../src/serialport_linux.cpp: In function 'int linuxGetSystemBaudRate(int, int*)':
../src/serialport_linux.cpp:29:16: error: 'TCGETS2' was not declared in this scope
   if(ioctl(fd, TCGETS2, &t)) {
                ^~~~~~~
make: *** [serialport.target.mk:98: Release/obj.target/serialport/src/serialport_linux.o] Error 1

### Steps and Code to Reproduce the Issue
npm install

reconbot, 2018-02-08:
Cc @fumon

Fumon, 2018-02-15:
Sorry for the delay, notification got filtered. Looks like an easy fix.

Seems to be an issue with musl libc which Alpine Linux is based on. Found a reference and fix in a similar  serial based project [here](https://github.com/tio/tio/pull/65) which appeared to resolve the issue.

I have a branch for you to try sticking in your package.json to see if it compiles https://github.com/Fumon/node-serialport/tree/fix_musl

If I don't hear from you @piechade, I'll spin up an Alpine VM to test this myself.

piechade, 2018-02-15:
@Fumon i will test it later this day. I need to look how to use GitHub instant of npm

Fumon, 2018-02-15:
Ah, you need
`npm install git+https://github.com/Fumon/node-serialport.git#fix_musl`
to get the correct branch.

I just built a docker image using:

```
FROM gliderlabs/alpine:3.4
RUN apk add --no-cache build-base linux-headers nodejs git python\
  && npm install git+https://github.com/Fumon/node-serialport.git#fix_musl
ENTRYPOINT /bin/sh
```

Which built successfully.

piechade, 2018-02-15:
Sorry about that, i forgot the branch. For me ist also everything working.

PR #1487 (2018-02-15): Alpine Linux: Adds missing header file for musl libc
Fixes #1470  After adding the new custom baud rate implementation for Linux, a build breaking bug occurred on Alpine Linux due to musl's handling of the kernel header files involved with ioctls. This patch introduces a fix by using the solution found [here](https://github.com/tio/tio/pull/65) on a similar, serial oriented project.

Tested in an Alpine Linux docker container. Passing all hardware tests.
commit 02e2bfec (2018-02-15): fix(linux): Adds missing header file for musl libc (eg alpine) (#1487)

Fixes #1470  After adding the new custom baud rate implementation for Linux, a build breaking bug occurred on Alpine Linux due to musl's handling of the kernel header files involved with ioctls. This patch introduces a fix by using the solution found [here](https://github.com/tio/tio/pull/65) on a similar, serial oriented project.

Tested in an Alpine Linux docker container. Passing all hardware tests.
closed 2018-02-15 by reconbot