serialport/node-serialport #1471: Not working on upgraded Node-Red on Raspberry PI 3/B

QAnders, 2018-02-08:
- __SerialPort Version__: 6.0.5
- __NodeJS Version__: v9.5.0
- __Operating System__ and __Hardware Platform__: Raspbian (latest) on R-Pi 3/B

- Have you checked the right __version__ of the __api docs?__: Think so
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
First of, Many thanks for your work!

This has been working before but I had to upgrade my Node-Red and Node.js versions to support AWS nodes for Node-Red.
I have spent a lot of time Google'ing and trying various different ways to install the serialport but it is still not working, sudo, non-sodu, global, etc.
Starting Node-Red on my RaspberryPi gives the following in the log:
`
8 Feb 05:53:14 - [info] Node-RED version: v0.18.2
8 Feb 05:53:14 - [info] Node.js  version: v4.2.1
8 Feb 05:53:14 - [info] Linux 4.9.59-v7+ arm LE
8 Feb 05:53:15 - [info] Loading palette nodes
8 Feb 05:53:18 - [info] Dashboard version 2.8.0 started at /ui
It looks like serialport didn't compile properly. This is a common problem and its fix is well documented here https://github.com/voodootikigod/node-serialport#to-install
The result of requiring the package is:  undefined
8 Feb 05:53:21 - [warn] ------------------------------------------------------
8 Feb 05:53:21 - [warn] [node-red-node-serialport/serialport] Error: Module version mismatch. Expected 46, got 59.
8 Feb 05:53:21 - [warn] [node-red-node-arduino/arduino] Missing serialport dependency
8 Feb 05:53:21 - [warn] ------------------------------------------------------
`

### Steps and Code to Reproduce the Issue

Upgrade Node.js and Node-Red to latest versions I guess...

QAnders, 2018-02-08:
I just did a new global install (over the previous global install) and now I think it went better. Created this little snippet:
```
var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyUSB1', {
  baudRate: 57600
});

// Switches the port into "flowing mode"
port.on('data', function (data) {
  console.log('Data:', data);
});

// Read data that is available but keep the stream from entering "flowing mode"
port.on('readable', function () {
  console.log('Data:', port.read());
});
```
It spews information out from my connected Ardiuno Nano:
```
Data: <Buffer e0 00 e0 e0 e0 00 e0 e0 00 e0 e0 e0 00 e0 00 00 00 e0 e0 e0 e0 00 e0 e0 e0 e0 e0 00 e0 e0 e0 00>
Data: <Buffer e0 00 00 e0 e0 00 e0 00 e0 e0 e0 e0 e0 e0 00 00 e0 e0 e0 00 e0 e0 00 e0 00 e0 00 00 e0 00 00 e0>
Data: <Buffer 00>
```

Thus it seems it is `node-red-node-serialport` that is the culprit for me... I will reference this Issue in Node-Red instead!

fivdi, 2018-02-09:
The error message shown is:
```
Error: Module version mismatch. Expected 46, got 59.
```
This means that serialport was installed/built with Node.js v9.x but at runtime Node.js v4.x is being used.

The `NODE_MODULE_VERSION` numbers for most releases of Node.js can be found at https://nodejs.org/en/download/releases/.

When native modules (containing C/C++ code)  like serialport are used, the NODE_MODULE_VERSION number of the Node.js versions used at install/build time must be the same as the NODE_MODULE_VERSION number of the Node.js version used at runtime.

I'll go ahead go close this issue as the [corresponding Node-RED issue](https://github.com/node-red/node-red-nodes/issues/404) has also been closed.

closed 2018-02-09 by fivdi

QAnders, 2018-02-09:
For anyone else running into this, nvm install was 9.5.0 but root user had 4.2.1 and since serialport had to be installed as root it was using the wrong version...
I removed nvm and upgraded both root and pi users Node.js versions and then it ran...