serialport/node-serialport #1472: Unable to get it working in Electron. (This version of Node.js requires NODE_MODULE_VERSION 57.)

MichMich, 2018-02-08:
- __SerialPort Version__: Latest
- __NodeJS Version__: v9.4.0
- __Operating System__ and __Hardware Platform__: Raspbian GNU/Linux 9 (stretch) - Raspberry Pi B2
- __Electron Version__: 1.8.2

- Have you checked the right __version__ of the __api docs?__: Yes.
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes.
- Are you using __Electron__ and have you checked the _Electron Docs_?: Yes.

### Summary of Problem

I'm trying to use Serialport within a Electron application. Serialport describes the process in the documentation, but no matter what I try, Serialport keeps giving me the following error.
>Error: The module '/home/pi/MagicMirror/modules/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 48. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

When running electron-rebuild, I get `✔ Rebuild Complete`. to be sure, I removed the build folder before running electron-rebuild, so it seems electron-rebuild is really doing it's work.

I've reinstalled all the node modules by removing the node_modules folder en rerunning `npm install`.

The last 12 hours have been nothing but googling, but so far I'm not able to solve the issue.

Any help is extremely appreciated!

### Steps and Code to Reproduce the Issue

- `npm install`
- `npm install electron-rebuild`
- `npm install serialport`
- `rm -Rf node_modules/serialport/build`
- `electron-rebuild`
- `npm run start`

MichMich, 2018-02-09:
I finally managed to get it to work!

What I did was:
- Remove node_modules folder from my project.
- Remove any global electron packages (electron & electron-rebuild)
- Remove any global Serialport packages
- Reboot (CRITICAL STEP!)
- Reinstall using the documentation of serial port.

So I think the reboot helped me.

closed 2018-02-09 by MichMich

reconbot, 2018-02-09:
yay! 👏