serialport/node-serialport #1474: API Docs unavailable for some browsers

morganrallen, 2018-02-09:
- __SerialPort Version__: All
- __NodeJS Version__: All
- __Operating System__ and __Hardware Platform__: All

- Have you checked the right __version__ of the __api docs?__: Unavailable
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Unavailable
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem
doclets.io SSL certificate has expired. According to the error it expired "January 27, 2018, 12:03 PM" but as issue from July 3, 2018 indicated this is an on going issue. https://github.com/lipp/doclets/issues/335

### Steps and Code to Reproduce the Issue
visit: https://doclets.io/node-serialport/node-serialport/
Firefox will not permit ignoring the error, Chrome/Brave appears to allow it. Haven't check others.

Maybe taking advantage of Github Pages is a more future proof way to move forward?

reconbot, 2018-02-09:
> is an on going issue. lipp/doclets#335

That sucks, it's kinda hard to hook it up to pages, but I'm for it.

---
Francis Gulotta
wizard@roborooter.com

morganrallen, 2018-02-09:
What do you see, off the top of your head, as the biggest pain point(s)?

reconbot, 2018-02-09:
How to build a different branch off of master. Npm run docs should make
good docs. Getting them on to GitHub pages is the part I don't know how to
solve.

IvanSanchez, 2018-02-14:
Have you considered using a `docs/` directory in the `master` branch? It's [what we use](https://github.com/Leaflet/Leaflet/tree/master/docs) in one of the projects I work on - it was the least painful thing to do.

reconbot, 2018-02-14:
I'm for it

reconbot, 2018-02-15:
Thank you for reporting this =) The docs will be up shortly at our github page https://node-serialport.github.io/node-serialport/?1

reconbot, 2018-02-15:
Also thank you for the idea =)

closed 2018-02-15 by reconbot

morganrallen, 2018-02-15:
Heck yeah! And totally ++ on the `doc/` directory tip, I hadn't been aware of that change.