serialport/node-serialport #1475: Node.js version in VS Code Extension

olback, 2018-02-12:
- __SerialPort Version__:6.0.5
- __NodeJS Version__:v8.9.4
- __Operating System__ and __Hardware Platform__: Antergos (Linux) 64-bit

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: N/A

### Summary of Problem
VS Code debugger throws versioning error when initializing plugin.
```
Node.js version using NODE_MODULE_VERSION 57. This version of Node.js requires NODE_MODULE_VERSION 54. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or`npm install`).
```

### Steps and Code to Reproduce the Issue
1. Make a new VS Code Extension Project with $`yo code`
2. `npm install serialport` or `npm install serialport --build-from-source`. 

```javascript
const serial = require('serialport');
```

### Other notes
I tried to fix the issue by doing the same as explained in #1472 with no luck.  
I also tried with `npm rebuild serialport --update-binary`, `npm rebuild serialport --build-from-source` and `npm rebuild` with no luck there either.

```
Activating extension `olback.vscode-serial` failed: The module '/home/olback/Desktop/vscode-serial/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`)..
Activating extension `olback.vscode-serial` failed:  The module '/home/olback/Desktop/vscode-serial/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
Here is the error stack:  Error: The module '/home/olback/Desktop/vscode-serial/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

```

hanfengcan, 2018-02-14:
I used the last electron version

reconbot, 2018-02-28:
You'll need to open an issue with vscode for this. They're not using the same node you're using in your cli. I unfortunately have no idea how they work.

closed 2018-02-28 by reconbot