serialport/node-serialport #1479: Raspberry read serial data from Arduino over USB and NodeJS

cassiolucas, 2018-02-14:
Hello guys, I would like to know if you could help me,

I am currently trying to configure the NodeJS on the Raspberry to read and store the values, currently the Arduino sends the following message over USB:

S1T1,2345,S1RSSI,-84,S1SNR,750

Where:
S1T1 = Means Sensor 1 Temperature 1;
2345 = Means 23,45 Celcius;
S1RSSI = Means Sensor 1 RSSI (Signal strenght);
-84 = Means the signal value -84;
S1SNR = Means Sensor 1 SNR (Signal strenght);
750 = Means 7.50 dBm.

As you can see below, I am able to receive the messages on Raspberry, but, I dont know how to allocate them into variables.

[image]

What I need is (I will try to describe as C++ as it works just fine in C++):

void loop()
{
  Blynk.run();
//***********************************************************************************//
// If there is serial data, read and store
  while (Serial.available() > 0) {
    
    if (Serial.find("S1T1")) {
    float t1 = Serial.parseInt();
    float temp1 = t1/100;
    Blynk.virtualWrite(V0, temp1);
//***********************************************************************************//
//Receive signal strength data
    if (Serial.find("S1RSSI")) {   
    int rssi1 = Serial.parseInt();
    Blynk.virtualWrite(V1, rssi1);
    }
    if (Serial.find("S1SNR")) {   
    float snr1 = Serial.parseInt();
    float snrs1 = snr1/100; 
    Blynk.virtualWrite(V2, snrs1);
    } 
  }

**The NodeJS code I have so far:**

const SerialPort = require('serialport');
const port = new SerialPort('/dev/ttyACM0', () => {
  console.log('Port Opened');
});
const parsers = SerialPort.parsers;

const parser = new parsers.Readline({
  delimiter: '\n'
 
});

port.pipe(parser);

parser.on('data', console.log);

**For reference, a Blynk example using virtual pins:**
http://help.blynk.cc/how-to-connect-different-hardware-with-blynk/raspberry-pi/how-to-install-nodejs-library-on-linux

var Blynk = require('blynk-library');
var AUTH = 'YourAuthToken';
var blynk = new Blynk.Blynk(AUTH);

var v1 = new blynk.VirtualPin(1);
var v9 = new blynk.VirtualPin(9);

v1.on('write', function(param) {
  console.log('V1:', param[0]);
});

v9.on('read', function() {
  v9.write(new Date().getSeconds());
});

Thank you.

copperlizard, 2018-02-17:
I'm just learning this stuff myself; but I saw your question and thought I'd try to be helpful...

in the JS code you have, the line "parser.on('data', console.log);" is using the parser to fire a callback function when the "data" event occurs. A "data" parameter (a buffer object to be specific) is automatically generated and passed to the callback function. So you could change your code to the something like...

function ReadSerialData(data){
  console.log(data);
  //do stuff here
}

const SerialPort = require('serialport');
const port = new SerialPort('/dev/ttyACM0', () => {
console.log('Port Opened');
});
const parsers = SerialPort.parsers;

const parser = new parsers.Readline({
delimiter: '\n'

});

port.pipe(parser);

parser.on('data', ReadSerialData);

... I think that's right; hope it helps.

littlej247, 2018-02-28:
I'm working on a similar project except the pi also runs a node web server that will display the outputs and runs a min console to send requests back. the format i use is "par=0/n" maybe reading it over can help you out. https://github.com/littlej247/ShowerPower/

closed 2018-11-22 by stale[bot]