serialport/node-serialport #1480: 'serialport' close event is not working properly

gauthamkantharaju, 2018-02-14:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: v4.4.4
- __Operating System__ and __Hardware Platform__: Linux on RaspberryPI

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
Disconnect or close event does not work properly !!
- close event is not trigerred
- application is working fine there after

Even tried using usb-detection npm package, no luck !!
### Steps and Code to Reproduce the Issue
Simple NodeJS application with open, close, error and data event functions is enough to reproduce the issue.

gauthamkantharaju, 2018-02-14:
Do we need to read or write functionality to detect removal? If not, how are going to detect the close event? Let me know.

Thank you in advance,
Gautham

IvanSanchez, 2018-02-15:
Have you tried with node-serialport version 6.x.x?

gauthamkantharaju, 2018-02-16:
No i haven't tried serialport version 6.x.x. Will definitely try and get back.

Do you want me to try any specific version? If so, please let me know the version because there is already an issue reported with 6.x.x version.

gauthamkantharaju, 2018-02-16:
https://github.com/node-serialport/node-serialport/issues/1465

gauthamkantharaju, 2018-02-16:
serialport v6.0.5 does not have a serialport.list() API for UNIX/Linux?

serialport.list(function(err, ports) {
		ports.forEach(function(port) {

reconbot, 2018-02-16:
It does

---
Francis Gulotta
wizard@roborooter.com

gauthamkantharaju, 2018-02-16:
Hi @reconbot,

Using below source code but unable to parse incoming data on "data" event. Let me know if I am missing something 

var serialport = require('serialport');
var portName = process.argv[2];

var myPort = new serialport(portName, {
    baudRate: 9600,
    parser: new serialport.parsers.Readline('\r\n')
});

myPort.on('open', onOpen);
myPort.on('data', onData);

reconbot, 2018-02-28:
Even with 5.0 parsers are transform streams and are must be piped to. 

From the [delimiter parsers docs](file:///Users/wizard/src/node-serialport/docs/DelimiterParser.html)
```
const SerialPort = require('serialport');
const Delimiter = SerialPort.parsers.Delimiter;
const port = new SerialPort('/dev/tty-usbserial1');
const parser = port.pipe(new Delimiter({ delimiter: '\n' }));
parser.on('data', console.log);
```

closed 2018-02-28 by reconbot