serialport/node-serialport #1481: Add status.dtr to modemBitsCallback

AgileDog, 2018-02-14:
- __SerialPort Version__: 6.1.0
- __NodeJS Version__: 8.9.3
- __Operating System__ and __Hardware Platform__: Windows 7 64 bit, Dell Precision T3500

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No trouble installing
- Are you using __Electron__ and have you checked the _Electron Docs_?: I don't know what this is.

### Summary of Problem
I am attempting to retrieve the status of the DTR line (high or low). There does not appear to be a way to do this. I looked through the docs and the code itself. With .set you can set true/false for BRK, CTS, DSR, DTR and RTS. For .get you only can check CTS, DSR and DCD (You cannot set DCD with .set). I would like to be able to check the status of DTR using .get. If there is another method for doing so, that would suffice.

### Steps and Code to Reproduce the Issue
This is just some test code, definitely not production code.
Note that DTR is set using .set
Then .get is used to retrieve settings. status will not contain information about DTR.

		var sbeSerialPort = new serialport(nameOfPort, configOfPort);
		sbeSerialPort.on('open', function ()
		{
			console.log("Connected to: ",sbeSerialPort.path);
			sbeSerialPort.set({dtr:false},function(){
				console.log('values set');
				sbeSerialPort.get(function(error, status){
					console.log('retrieved');
					console.log(error);
					console.log(status);
				});
			});
		});

reconbot, 2018-09-23:
I'm sorry my previous comment is incorrect.

reconbot, 2018-09-23:
DCD and DTR are linked. The DTR of your computer is linked to the DCD of the other device. And vice versa.  Because of this DTR is a write only attribute and DCD is a read only attribute. Getting the current status of DTR is something that should be possible. So I'll keep this issue open for that.