serialport/node-serialport #1489: npm update install pre relase

piechade, 2018-02-16:
- __SerialPort Version__: 6.0.5
- __NodeJS Version__: v9.4.0
- __Operating System__ and __Hardware Platform__: Windows 10 x64

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem

If i have the last stable installed and run npm update the pre relase version will be installed.

### Steps and Code to Reproduce the Issue

install 6.0.5
`npm install serialport@6.0.5 --save`
run update
`npm update

reconbot, 2018-02-16:
Latest and beta tags are correct not sure what to do.

piechade, 2018-02-17:
Tagged Releases
To publish a prerelease without making it installed by default, you should specify a tag other than “latest”. For example, to set the “next” tag:

npm publish --tag next

reconbot, 2018-02-28:
This is a known bug in npm upgrade

closed 2018-02-28 by reconbot