serialport/node-serialport #1490: Is there anyway to identify the originating port(pnpid) for a data event?

copperlizard, 2018-02-16:
- __SerialPort Version__: v6.0.4
- __NodeJS Version__: v6.11.3
- __Operating System__ and __Hardware Platform__: Ubuntu Intel Nuc

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem

I have multiple units talking to the server, and I would like my app to recognize which unit has been disconnected; all data events include a unit identifier. If I could associate a port pnpID with a unit id, then I could track disconnected units by passing the port pnpID with the "close" event (once the server hears from a unit on a particular port, it shows as connected, and if that port closes, that unit shows as disconnected). 

I would like the user to have the option to move usb cables around at their convenience, but still recognize when a particular unit is no longer connected. Any advice would be appreciated.

Event: "data"
The data event puts the port in flowing mode. Data is emitted as soon as it's received. Data is a Buffer object with a varying amount of data in it. The readLine parser converts the data into string lines. See the parsers section for more information on parsers, and the Node.js stream documentation for more information on the data event.

### Steps and Code to Reproduce the Issue
N/A

reconbot, 2018-02-28:
In a data event's call back `this` is bound to the port that caused the event. You can get `this.path` which is the com port. `Serialport.list()` can give more information about each port but the port itself doesn't have any of the extra info.

copperlizard, 2018-02-28:
Oh cool. Thanks.

closed 2018-02-28 by copperlizard