serialport/node-serialport #1491: Problem with the output

SpicyChef, 2018-02-17:
- __SerialPort Version__:  
- __NodeJS Version__: 8.9.4 LTS
- __Operating System__ and __Hardware Platform__: Mac OS, MacBook Pro

### Summary of Problem
I'm trying to display a message coming from my Arduino in the terminal.
I got the message to show up in the terminal but some times it shows up in a weird way like this
"
Port Opened
message written
Output: Hello

Output: Hello

Output: Hell
Output: o

Output: Hello

Output: Hello
Output: 

Output: Hello

Output: Hello

Output: Hel
Output: lo

Output: Hello

Output: Hello
Output: 
"

### Steps and Code to Reproduce the Issue
Arduino code:

```
void setup() {

  Serial.begin(9600);
}

void loop() {

  Serial.println("Hello");
  delay(1000);
  
}
```

Javascript-serialport code:
```
'use strict';

// Open event example
const SerialPort = require('serialport');
const port = new SerialPort('/dev/cu.usbserial-A505BRMT');
baudRate: 9600

port.on('open', () => {
  console.log('Port Opened');
});

port.write('main screen turn on', (err) => {
  if (err) { return console.log('Error: ', err.message) }
  console.log('message written');
});

port.on('data', (data) => {
	
  /* get a buffer of data from the serial port */
  console.log('Output: ' + data);
});

```

reconbot, 2018-02-17:
Consider using the new line parser. Data will be emitted as it is read from
the system.

SpicyChef, 2018-02-18:
I'm new to this, would you give me an example?

RyanAtRing, 2018-02-18:
@SpicyChef try this

`var SerialPort = require('serialport')
let Ready = SerialPort.parsers.Ready;
var port = new SerialPort('/dev/PATH_TO_USB', {
    baudRate: 9600,
})

let parser = port.pipe(new SerialPort.parsers.Readline);

parser.on('data', function(data) {
    console.log('Data: ', data);
})`

RyanAtRing, 2018-02-18:
@SpicyChef here's an example

```var SerialPort = require('serialport')
let Ready = SerialPort.parsers.Ready;
var port = new SerialPort('/dev/PATH_TO_USB', {
    baudRate: 9600,
})

let parser = port.pipe(new SerialPort.parsers.Readline);

parser.on('data', function(data) {
    console.log('Data: ', data);
})

RyanNelwan, 2018-02-18:
@SpicyChef here's an example

```
var SerialPort = require('serialport')
let Ready = SerialPort.parsers.Ready
var port = new SerialPort('/dev/PATH_TO_USB', {
    baudRate: 9600,
})

let parser = port.pipe(new SerialPort.parsers.Readline)

parser.on('data', function(data) {
    console.log('Data: ', data)
})

SpicyChef, 2018-02-19:
It worked, appreciate it.

closed 2018-02-19 by SpicyChef