serialport/node-serialport #1495: Serialport 6.1.0 won't install or build on Rpi3

rjbaat, 2018-02-20:
- __SerialPort Version__: 6.1.0
- __Operating System__ and __Hardware Platform__: Resin.io Rpi3

### Summary of Problem

When i install node-red-contrib-modbus and or node-red-node-serialport. I get the error that dependency serialport 6.1.0 has no prebuild version. But it also won't build.

```
[34m[Build]   [39m make: Leaving directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/node-modbus/node_modules/serialport/build'
[34m[Build]   [39m [91mprebuild-install[0m[91m [0m[91minfo[0m[91m begin Prebuild-install version 2.5.1
[34m[Build]   [39m [0m[91mprebuild-install info[0m[91m looking for local prebuild @[0m[91m prebuilds/serialport-v6.1.0-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install info[0m[91m looking for cached prebuild @[0m[91m /root/.npm/_prebuilds/https-github.com-node-serialport-node-serialport-releases-download-v6.1.0-serialport-v6.1.0-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install http [0m[91mrequest[0m[91m GET https://github.com/node-serialport/node-serialport/releases/download/v6.1.0/serialport-v6.1.0-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install http [0m[91m404 https://github.com/node-serialport/node-serialport/releases/download/v6.1.0/serialport-v6.1.0-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install WARN [0m[91minstall[0m[91m No prebuilt binaries found (target=9.4.0 runtime=node arch=arm platform=linux)
[34m[Build]   [39m [0mmake: Entering directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/poller.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[34m[Build]   [39m [91m../src/serialport_linux.cpp: In function 'int linuxSetCustomBaudRate(int, unsigned int)':
[34m[Build]   [39m ../src/serialport_linux.cpp:10:18: error: 'TCGETS2' was not declared in this scope
[34m[Build]   [39m      if(ioctl(fd, TCGETS2, &t)) {
[34m[Build]   [39m                   ^~~~~~~
[34m[Build]   [39m [0m[91m../src/serialport_linux.cpp:18:18: error: 'TCSETS2' was not declared in this scope
[34m[Build]   [39m      if(ioctl(fd, TCSETS2, &t)) {
[34m[Build]   [39m                   ^~~~~~~
[34m[Build]   [39m [0m[91m../src/serialport_linux.cpp: In function 'int linuxGetSystemBaudRate(int, int*)':
[34m[Build]   [39m ../src/serialport_linux.cpp:29:16: error: 'TCGETS2' was not declared in this scope
[34m[Build]   [39m    if(ioctl(fd, TCGETS2, &t)) {
[34m[Build]   [39m                 ^~~~~~~
[34m[Build]   [39m [0m[91mmake: *** [serialport.target.mk:96: Release/obj.target/serialport/src/serialport_linux.o] Error 1
[34m[Build]   [39m make: *** Waiting for unfinished jobs....
[34m[Build]   [39m [0mmake: Leaving directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
```
When i manually install npm serialport 6.0.5. it will build and install on the Rpi3 on Resin.io. But due to the dependency in the 2 nodes mentioned before it will try to install 6.1.0 also after 6.0.5 is installed.

```
[34m[Build]   [39m /usr/local/bin/serialport-list -> /usr/local/lib/node_modules/serialport/bin/list.js
[34m[Build]   [39m /usr/local/bin/serialport-repl -> /usr/local/lib/node_modules/serialport/bin/repl.js
[34m[Build]   [39m /usr/local/bin/serialport-term -> /usr/local/lib/node_modules/serialport/bin/terminal.js
[34m[Build]   [39m [91mprebuild-install[0m[91m [0m[91minfo [0m[91mbegin[0m[91m Prebuild-install version 2.5.1
[34m[Build]   [39m [0m[91mprebuild-install info [0m[91mlooking for local prebuild @ prebuilds/serialport-v6.0.5-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install info [0m[91mlooking for cached prebuild @ /root/.npm/_prebuilds/https-github.com-node-serialport-node-serialport-releases-download-v6.0.5-serialport-v6.0.5-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install http [0m[91mrequest[0m[91m GET https://github.com/node-serialport/node-serialport/releases/download/v6.0.5/serialport-v6.0.5-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install http [0m[91m404 https://github.com/node-serialport/node-serialport/releases/download/v6.0.5/serialport-v6.0.5-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install WARN install[0m[91m No prebuilt binaries found (target=9.4.0 runtime=node arch=arm platform=linux)
[34m[Build]   [39m [0mmake: Entering directory '/usr/local/lib/node_modules/serialport/build'
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/poller.o
[34m[Build]   [39m   SOLINK_MODULE(target) Release/obj.target/serialport.node
[34m[Build]   [39m   COPY Release/serialport.node
[34m[Build]   [39m make: Leaving directory '/usr/local/lib/node_modules/serialport/build'
[34m[Build]   [39m + serialport@6.0.5
[34m[Build]   [39m added 180 packages in 13.65s

```
### Steps and Code to Reproduce the Issue

Install node-red-node-serialport and/or node-red-contrib-modbus on Rpi3 with Resin.io docker file:

```
# base-image for node on any machine using a template variable,
# see more about dockerfile templates here:http://docs.resin.io/pages/deployment/docker-templates
# Note the node:slim image doesn't have node-gyp
FROM resin/%%RESIN_MACHINE_NAME%%-alpine-node:9-slim

### Defines our working directory in container
RUN mkdir -p /usr/src/app/
WORKDIR /usr/src/app

### This will copy all files in our root to the working  directory in the container
COPY ./app ./

### Install node modules, including i2c-bus without carrying all the deps in the layers
RUN apk add --no-cache make gcc g++ python && \
  JOBS=MAX npm install -g node-red node-red-contrib-resinio --production --silent && \
  JOBS=MAX npm install -g node-red-contrib-modbus --production --silent && \
  JOBS=MAX npm install -g node-red-node-serialport --production --silent && \
  apk del make gcc g++ python && \
  npm cache clean --force && rm -rf /tmp/*

### server.js will run when container starts up on the device
CMD ["bash", "/usr/src/app/start.sh"]

```

yutter, 2018-02-20:
I am seeing similar issues with node-red-contrib-modbus. I am using alpine slim with node v8.9.4

yutter, 2018-02-20:
This is a copy of [#1487](https://github.com/node-serialport/node-serialport/pull/1487) which has been fixed but not yet made a release

rjbaat, 2018-02-21:
Oke thnx. 
Is there an estimation when the fix will be released?

piechade, 2018-02-21:
Why you are using the pre-release version?

yutter, 2018-02-21:
@piechade it is included in a Node-RED library by someone else. We did not choose the version

reconbot, 2018-02-28:
Serialport 6.1.1 is out which should fix this. I see node-red has 6.0.5 as it's version, I have no idea why 6.1.0 would have installed.

reconbot, 2018-02-28:
Do let me know if this works for you.

closed 2018-02-28 by reconbot

rjbaat, 2018-02-28:
The developer of the Node Red module had 6.1.0 included. I asked him to downgrade to the stable version of 6.0.5 and that works. So i hope the node red module is staying on stable release from now on and then it should work fine.

reconbot, 2018-02-28:
Ask them to upgrade to 6.1.1 it should work better. Feel free to cc me, I
had asked them to test 6.1.0 I I didn't know they'd release it. 🙊

rjbaat, 2018-08-03:
i had to rebuild my docker on rpi again and i run into the same problem again. It won't build serialport.
i tried both separate node-serialport as node-red-contrib-modbus. with the command:

npm install -g node-red-contrib-modbus --production --silent
npm install -g node-red-node-serialport --production --silent

Both giving me the problem with building serialport. Any idea what the problem would be? I see in node-red-contrib-modbus its using 6.1.1. 

https://github.com/biancode/node-red-contrib-modbus/blob/ac65a11b15a50728b90474bc48726a036008e716/package.json#L12

and in node-red-node-modbus its using 6.2.0 if I am right. 

https://github.com/node-red/node-red-nodes/blob/50fc196c6f8d689ac82052d62f1065e6c5924ae6/io/serialport/package.json#L6

```
34m[main][39m  Step 7/9 : RUN apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*
[34m[main][39m   ---> Running in 25b061d23a60
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/armhf/APKINDEX.tar.gz
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/armhf/APKINDEX.tar.gz
[34m[main][39m  (1/13) Installing binutils (2.30-r5)
[34m[main][39m  (2/13) Installing isl (0.18-r0)
[34m[main][39m  (3/13) Installing libgomp (6.4.0-r8)
[34m[main][39m  (4/13) Installing libatomic (6.4.0-r8)
[34m[main][39m  (5/13) Installing mpfr3 (3.1.5-r1)
[34m[main][39m  (6/13) Installing mpc1 (1.0.3-r1)
[34m[main][39m  (7/13) Installing gcc (6.4.0-r8)
[34m[main][39m  (8/13) Installing musl-dev (1.1.19-r10)
[34m[main][39m  (9/13) Installing libc-dev (0.7.1-r0)
[34m[main][39m  (10/13) Installing g++ (6.4.0-r8)
[34m[main][39m  (11/13) Installing make (4.2.1-r2)
[34m[main][39m  (12/13) Installing gdbm (1.13-r1)
[34m[main][39m  (13/13) Installing python2 (2.7.15-r0)
[34m[main][39m  Executing busybox-1.28.4-r0.trigger
[34m[main][39m  OK: 205 MiB in 83 packages
[34m[main][39m  make: Entering directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/poller.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[34m[main][39m  [91m../src/serialport_linux.cpp:4:24: fatal error: asm/ioctls.h: No such file or directory
[34m[main][39m   #include <asm/ioctls.h>
[34m[main][39m                          ^
[34m[main][39m  compilation terminated.
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:96: Release/obj.target/serialport/src/serialport_linux.o] Error 1
[34m[main][39m  make: *** Waiting for unfinished jobs....
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport_unix.cpp:26:26: fatal error: linux/serial.h: No such file or directory
[34m[main][39m   #include <linux/serial.h>
[34m[main][39m                            ^
[34m[main][39m  compilation terminated.
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:96: Release/obj.target/serialport/src/serialport_unix.o] Error 1
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In static member function 'static void Poller::onData(uv_poll_t*, int, int)':
[34m[main][39m  ../src/poller.cpp:69:29: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     obj->callback.Call(2, argv);
[34m[main][39m                               ^
[34m[main][39m  In file included from ../src/poller.cpp:1:0:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterOpen(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:95:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(2, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[... 51 lines cut ...]
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterGetBaudRate(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:383:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(2, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterDrain(uv_work_t*)':
[... cut ...]
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  make: Leaving directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
[34m[main][39m  Removing intermediate container 25b061d23a60
```

rjbaat, 2018-08-03:
I have also tried with a newer resin alpine base img with newer nodejs version 0.10.48 but that gives me the same errors:

https://github.com/resin-io-library/base-images/blob/5159250f897aa4e88c701d4c27dac62645bafcaa/node/raspberry-pi/alpine/0.10/Dockerfile

```
[34m[main][39m  Step 1/9 : FROM resin/raspberrypi3-alpine-node:0.10.48-slim
[34m[main][39m   ---> 46a9c2b36740
[34m[main][39m  Step 2/9 : RUN mkdir -p /usr/src/app/
[34m[main][39m   ---> Running in fb26500e3558
[34m[main][39m   ---> ceb6eceab279
[34m[main][39m  Removing intermediate container fb26500e3558
[34m[main][39m  Step 3/9 : WORKDIR /usr/src/app
[34m[main][39m   ---> 2a0412617ae2
[34m[main][39m  Removing intermediate container fe8976e19575
[34m[main][39m  Step 4/9 : COPY ./app ./
[34m[main][39m   ---> 3de0661066b8
[34m[main][39m  Removing intermediate container be9cde7470b2
[34m[main][39m  Step 5/9 : COPY entry.sh /usr/bin/entry.sh
[34m[main][39m   ---> 4e53fa4dbf33
[34m[main][39m  Removing intermediate container ae3d549d9826
[34m[main][39m  Step 6/9 : COPY udev-rules/ /etc/udev/rules.d/
[34m[main][39m   ---> 6f40895ec386
[34m[main][39m  Removing intermediate container ba14080f9d1a
[34m[main][39m  Step 7/9 : RUN apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-node-serialport --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*
[34m[main][39m   ---> Running in 47854f1ebb21
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/armhf/APKINDEX.tar.gz
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/armhf/APKINDEX.tar.gz
[34m[main][39m  (1/13) Installing binutils (2.30-r5)
[34m[main][39m  (2/13) Installing isl (0.18-r0)
[34m[main][39m  (3/13) Installing libgomp (6.4.0-r8)
[34m[main][39m  (4/13) Installing libatomic (6.4.0-r8)
[34m[main][39m  (5/13) Installing mpfr3 (3.1.5-r1)
[34m[main][39m  (6/13) Installing mpc1 (1.0.3-r1)
[34m[main][39m  (7/13) Installing gcc (6.4.0-r8)
[34m[main][39m  (8/13) Installing musl-dev (1.1.19-r10)
[34m[main][39m  (9/13) Installing libc-dev (0.7.1-r0)
[34m[main][39m  (10/13) Installing g++ (6.4.0-r8)
[34m[main][39m  (11/13) Installing make (4.2.1-r2)
[34m[main][39m  (12/13) Installing gdbm (1.13-r1)
[34m[main][39m  (13/13) Installing python2 (2.7.15-r0)
[34m[main][39m  Executing busybox-1.28.4-r0.trigger
[34m[main][39m  OK: 205 MiB in 83 packages
[34m[main][39m  [91m
[34m[main][39m  [0m
[34m[main][39m  [91m/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/prebuild-install/node_modules/which-pm-runs/index.js:11
[34m[main][39m  [0m
[34m[main][39m  [91m  const pmSpec = userAgent.split(' ')[0]
[34m[main][39m    ^^^^^
[34m[main][39m  [0m
[34m[main][39m  [91mSyntaxError: Use of const in strict mode.
[34m[main][39m      at Module._compile (module.js:439:25)
[34m[main][39m      at Object.Module._extensions..js (module.js:474:10)
[34m[main][39m      at Module.load (module.js:356:32)
[34m[main][39m      at Function.Module._load (module.js:312:12)
[34m[main][39m      at Module.require (module.js:364:17)
[34m[main][39m      at require (module.js:380:17)
[34m[main][39m      at Object.<anonymous> (/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/prebuild-install/bin.js:5:19)
[34m[main][39m      at Module._compile (module.js:456:26)
[34m[main][39m      at Object.Module._extensions..js (module.js:474:10)
[34m[main][39m      at Module.load (module.js:356:32)
[34m[main][39m  [0m
[34m[main][39m  make: Entering directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/poller.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[34m[main][39m  [91m../src/serialport_linux.cpp:4:24: fatal error: asm/ioctls.h: No such file or directory
[34m[main][39m   #include <asm/ioctls.h>
[34m[main][39m                          ^
[34m[main][39m  compilation terminated.
[... 106 lines cut ...]
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterDrain(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:424:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(1, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../node_modules/nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:91: Release/obj.target/serialport/src/poller.o] Error 1
[34m[main][39m  [0m
[34m[main][39m  make: Leaving directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[... cut ...]
[================>                                  ] 33%[2K[G[32m[Success][39m  Successfully uploaded images
[31m[Error][39m  Some services failed to build:
[31m[Error][39m    Service: main
[31m[Error][39m      Error: The command '/bin/sh -c apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-node-serialport --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*' returned a non-zero code: 1
[31m[Error][39m  Not deploying release.
```