serialport/node-serialport #1496: memory leak when continuously writing?

jsyzqt, 2018-02-20:
- __SerialPort Version__: 6.0.4 & 6.0.5 & 6.1.0
- __NodeJS Version__: v8.9.0
- __Operating System__ and __Hardware Platform__: Windows 10 x64 [10.0.16299.248]

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:
First used in Electron, and then test in a pure node environment after encountering the problem.

### Summary of Problem
I experienced that when continuously writing data to the serial port, large numbers of 'Buffer' objects were created, which can be observed in DevTools using 'Take snapshot'. If the transmit speed was accelerated, sometimes the programs will occupy hundreds of MBs of memory in a daytime. I have tried the versions of 6.0.4 & 6.0.5 & 6.1.0.
### Steps and Code to Reproduce the Issue
I simply use the following test codes.
```
var port = new SerialPort("COM20", {
    baudRate: defaultBaudRate
});
setTimeout(()=>{
    setInterval(() => {
        port.write(Buffer.from('##0115aC8**', 'ascii'));
    }, 300);

    port.on('data', function(data){
        console.log(data.toString('ascii'));
    });
}, 400);
```

reconbot, 2018-02-20:
You're not waiting for the writes to finish so they probably just queue.

jsyzqt, 2018-02-20:
@reconbot There are enough time intervals between sending intructions. Actually, the response data is correct and in time. My arduino reply instructions every 50ms, so I think the time interval should be enough.

reconbot, 2018-02-28:
If you could maybe setTimeout after the write just to be sure that would help you debug.

jsyzqt, 2018-03-02:
@reconbot Recently, I have tried the following code:
```
const promisify = require('util').promisify;
const SerialPort = require('serialport');

var port = new SerialPort("COM20", {
    baudRate: 115200
});

port.on('data', function(data){
    console.log(data.toString('ascii'));
});

(function test(){
    promisify(wAd)().then(()=>{
        test();
    });
})();

function wAd(callback){
    port.write(Buffer.from('##0115aC8**', 'ascii'));
    port.drain((err)=>{
        setTimeout(()=>{
            callback();
        }, 1000);
    });
}
```
However, the problem is still same as what I have previously said. The 'buffer' objects are accumulating after starting the program. But, I find it works well on Ubuntu 16.04 and the memory usage is stable.

benpurdy, 2018-04-09:
I have seen this same problem, also using Windows. The "rss" memory reported from node will steadily climb as the serial port is used (I need to write to the port around 40 times per second the entire time the app is running, and the app needs to run for many days).

I have been able to fix the issue by adding a single line to `serialport_win.cpp`

Please note that I'm _extremely unfamiliar_ with native code binding in node but from what I can understand in the documentation for nan::Persistent, it seems like V8 might be keeping the underlying memory around even though the baton has been deleted since `baton->buffer.Reset(buffer)` is called but `baton->buffer.Empty()` or `baton->buffer.Reset()` aren't called later.

Anyhow, adding `baton.buffer.Reset()` just before `delete baton;` on line #387 in `serialport_win.cpp` appears to resolve this problem on my machine.

reconbot, 2018-04-09:
That's interesting. It's a [different line in master](https://github.com/node-serialport/node-serialport/blob/master/src/serialport_win.cpp#L391) the `AFTER_WRITE()` right? This is [the only place](hthttps://github.com/node-serialport/node-serialport/blob/9dbccbc755e622b1c3d625625af4c627cc98d432/src/serialport_win.h#L20) we keep a buffer any longer. (We used to do this for all systems.) I'm not familiar enough to know if we're freeing it correctly. I think `reset()` is probably setting it to be 0 length which is freeing most of it but probably not the object itself.

@indutny do you know off the top of your head?

benpurdy, 2018-04-09:
Oh, I was just looking at the source again, would it be better to dispose of the buffer in the [WriteIOCompletion ](https://github.com/node-serialport/node-serialport/blob/9dbccbc755e622b1c3d625625af4c627cc98d432/src/serialport_win.cpp#L345) function?

I don't know if this is a problem or not but my previous solution will release the buffer as soon as the write is queued (via EIO_AfterWrite) rather than when the write operation is actually _finished_..

reconbot, 2018-04-09:
I'll double check but I don't think we call after write until it's
completed.

PR #1547 (2018-04-10): baton buffer is now reset after write operations are complete.
Resetting the buffer before deleting the struct appears to resolve a memory leak being reported by Windows users.

closes #1496
closed 2018-04-10 by reconbot