serialport/node-serialport #1498: serial port based barcode scanner

cissemy, 2018-02-23:
- __SerialPort Version__:
- __NodeJS Version__:
- __Operating System__ and __Hardware Platform__: windows

How can I get data from serial port based barcode scanner using  node-serialport ?
Thanks

fivdi, 2018-02-24:
@cissemy Manufacturers of barcode scanners typically provide documentation for their scanners in the form of a user guide or reference manual. This documentation should describe the protocol which can be used to communicate with the scanner. With this information it should be possible to write a program that uses serialport to communicate with the scanner.

reconbot, 2018-02-25:
You got to give some example code or something.

reconbot, 2018-02-28:
Let us know if you can give us more info.

closed 2018-02-28 by reconbot