serialport/node-serialport #1502: Opening COM3: Access Denied

nickabond, 2018-02-27:
- __SerialPort Version__:
6.0.5

- __NodeJS Version__:
8.2.1

- __Operating System__ and __Hardware Platform__:

Windows 10, 64 bit

- Have you checked the right __version__ of the __api docs?__:
Yes

- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
Installation went fine.

- Are you using __Electron__ and have you checked the _Electron Docs_?:
I am using Electron.

### Summary of Problem

I'm getting an Error: Opening COM3: Access Denied error when I'm trying to open a port. I'm using a laboratory scale, and it seems to work the first time I open my program. However, if the scale is unplugged or if I close my program and restart it while the scale is still plugged in I get the Access Denied error when trying to open the port on the subsequent time.

It seems like it isn't telling the computer that I'm no longer using the port when it tries to access the port on the second time. 

Also, I'm not sure if it is connected but when the port does open successfully it reads data for a time, but eventually the "data" event stops firing for some reason?

Any ideas?

### Steps and Code to Reproduce the Issue

I'm using the 6.0.5 version of Serial Port and I am using Electron Version 1.8.2

nickabond, 2018-02-27:
Here's my code:

`
 attachHandlers:function(){

        this.port.on('data', function (data) {

            for(var i = 0;i<data.length;i++){
              
              if(data[i] != 32){
    
                finalString += String.fromCharCode(data[i]);
    
              }
            }
            
            if(finalString.indexOf("g") > -1){
              this.currentReading = parseFloat(finalString);
              console.log(this.currentReading);
              finalString = "";
            }
            
          });

        this.port.on('close', function(){
            console.log("DISCONNECTED");
        });

    },
connectScale:function(){

        this.port = new SerialPort('COM3', {baudRate:9600,dataBits: 8,stopBits: 1},function (err) {
            if (err) {
              //Scale.connectScale();
              return console.log('Error: ', err.message);
            
            }
            else{
              console.log("IT worked.");
              Scale.attachHandlers();
            }
        });

        

    }`

nickabond, 2018-02-27:
The connectScale function only gets called once when the Electron application starts.

nickabond, 2018-02-28:
Actually it looks like I solved the issue!

For one it seemed to be an issue when I opened the port automatically. Once I opened the port manually, and closed the port on the 'will-quit' and 'quit' electron app events then everything seemed to work fine. It also looks like it took care of the issue where the scale was only receiving data events for 15 minutes or so.

Long story short....open the port manually, and make sure to close it because otherwise Windows still thinks it is in use.

nickabond, 2018-02-28:
To help others here is my code to open the port now:
`this.port = new SerialPort('COM3', {autoOpen:false,baudRate:9600,dataBits: 8,stopBits: 1});

        this.port.open(function(err){
             
            if(err){
                alert("Failed to open port.");
            }
            else{

                document.getElementById("COMPortStatus").innerText = "COM3 Is Open"
                Scale.attachPortHandlers();

            }

        });`

And here is the code to run before the app quits. In Electron this went in my main.js

`
app.on('will-quit',function(){

  Scale.port.close();

});

app.on('quit',function(){

  Scale.port.close();
  
});`

Hope that helps someone else!

closed 2018-03-01 by nickabond

reconbot, 2018-03-01:
This is so weird, I'm glad you worked it out.