serialport/node-serialport #1506: Problem to receive HEX Data

wendy2702, 2018-02-28:
- __SerialPort Version__:6.1.1
- __NodeJS Version__: v6.13.0
- __Operating System__ and __Hardware Platform__: RPI3 Linux Jessie

Hi all,

i'm quite new in Javascript and git so apologize for any mistakes i made opening this issue.

 #I try to read data from a Heating device but i failed to receive the data via script. From Serial port sniffer i can see that the answer is coming but for some reason i did not see them at i.e. my log/console.

This is the script i used:

```var SerialPort = require("serialport");
var buffer = new Buffer(12);
buffer[0] = 0x10;
buffer[1] = 0x02;
buffer[2] = 0x01;
buffer[3] = 0x15;
buffer[4] = 0x00;
buffer[5] = 0x00;
buffer[6] = 0x00;
buffer[7] = 0xf3;
buffer[8] = 0x10;
buffer[9] = 0x03;
buffer[10] = 0x7c;
buffer[11] = 0x32;
var port = new SerialPort("/dev/ttyUSB0", {
baudRate: 9600,
dataBits : 8,
parity : 'none',
stopBits: 1,
flowControl : false
},
function (err) {
if (err) {
return console.log('Error1: ', err.message);

}

port.on('data', function (data) {
console.log('Data: ' + data);
var daten=[];
daten=data.toString('hex');
setState('Waterkotte.Daten.Rohdaten', daten & 0xFF,true);
port.close();
});

// Flush input
port.flush();

// send hello
port.write(buffer, function(err) {
if (err) {
return console.log('Error on write: ', err.message);

}
console.log('Daten gesendet');
// port.close();
});
});

port.on('open', function() {
console.log('Connected');
});

port.on('error', function(err) {
console.log('Error2: ', err.message);
// port.close();
});

// close port if the script stopped (to be able to open it again)
onStop(function (callback) {
if (port && port.isOpen) {
port.close();
console.log('port closed');
}
// callback();

});
```

If i use the serial port sniffer tool: jpnevulator which was startet with:

`jpnevulator --ascii --timing-print --tty /dev/ttyUSB0:SB9600d --read`

I can see that the correct answer is coming:

```
2018-02-28 10:14:54.147334:
FF 10 02 00 17 1F BE 12 0C 06 03 05 29 11 C0 D3 ............)...
47 4E C2 C0 C0 15 DD C0 C8 E6 D8 41 13 53 E1 41 GN.........A.S.A
9D B8 09 42 D3 FF 33 42 69 7D 27 42 52 38 DE C2 ...B..3Bi}'BR8..
00 00 70 41 12 58 86 BE B0 D9 91 C0 E2 39 13 C1 ..pA.X.......9..
2E 59 13 42 66 66 86 C0 00 00 00 00 ED FE 87 3F .Y.Bff.........?
47 F5 04 41 C8 E6 D8 41 13 53 E1 41 00 00 90 41 G..A...A.S.A...A
13 53 E1 41 69 7D 27 42 10 10 0B 0B 1C 02 12 96 .S.Ai}'B........
03 5D 46 DE C9 1B 46 1B FD 82 45 00 00 00 00 01 .]F...F...E.....
01 01 01 01 01 01 01 01 01 01 01 00 00 00 00 00 ................
C7 53 00 00 02 09 25 2A 12 12 01 0D 01 E3 20 00 .S....%*...... .
00 4E 5B 1D C0 82 E9 43 40 92 E6 17 40 E1 51 FE .N[....C@...@.Q.
3E 09 6F 14 42 B5 8B 00 42 CC 77 1B 41 3F 3F 47 >.o.B...B.w.A??G
42 02 00 00 00 A0 41 00 00 00 00 00 00 00 00 2C B.....A........,
ED 02 00 02 A3 20 00 20 00 00 00 02 00 00 00 00 ..... . ........
00 00 00 03 01 08 08 00 00 00 00 00 00 00 01 27 ...............'
00 7D 01 27 00 28 35 12 10 03 19 DF FF          .}.'.(5......
```
But this Hex string i'm not able to see as "data" at my console/log.

This is an example feedback:

```
14:00:41.247   [info]   javascript.0 Stop script script.js.Skript1
14:00:41.277   [info]   javascript.0 Start javascript script.js.Skript1
14:00:41.278   [info]   javascript.0 script.js.Skript1: registered 0 subscriptions and 0 schedules
14:00:41.279   [info]   javascript.0 script.js.Skript1: Connected
14:00:41.279   [info]   javascript.0 script.js.Skript1: Daten gesendet
14:00:41.391   [info]   javascript.0 script.js.Skript1: Data: d0
14:00:54.192   [info]   javascript.0 Stop script script.js.Skript1
14:11:15.612   [info]   javascript.0 Stop script script.js.Skript1
14:11:18.807   [info]   javascript.0 Start javascript script.js.Skript1
14:11:18.808   [info]   javascript.0 script.js.Skript1: registered 0 subscriptions and 0 schedules
14:11:18.814   [info]   javascript.0 script.js.Skript1: Connected
14:11:18.818   [info]   javascript.0 script.js.Skript1: Daten gesendet
14:11:18.843   [info]   javascript.0 script.js.Skript1: Data: 
14:11:28.669   [info]   javascript.0 Stop script script.js.Skript1
14:11:29.698   [info]   javascript.0 Start javascript script.js.Skript1
14:11:29.699   [info]   javascript.0 script.js.Skript1: registered 0 subscriptions and 0 schedules
14:11:29.700   [info]   javascript.0 script.js.Skript1: Connected
14:11:29.700   [info]   javascript.0 script.js.Skript1: Daten gesendet
14:11:29.753   [info]   javascript.0 script.js.Skript1: Data: @
14:11:45.748   [info]   javascript.0 Stop script script.js.Skript1
14:11:47.483   [info]   javascript.0 Start javascript script.js.Skript1
14:11:47.485   [info]   javascript.0 script.js.Skript1: registered 0 subscriptions and 0 schedules
14:11:47.485   [info]   javascript.0 script.js.Skript1: Connected
14:11:47.485   [info]   javascript.0 script.js.Skript1: Daten gesendet

As you can see the value for "Data" is different all the time.

What i'm doing wrong?

Thanks for help

IvanSanchez, 2018-02-28:
This piece of code here...

```
port.on('data', function (data) {
  console.log('Data: ' + data);
  var daten=[];
  daten=data.toString('hex');
  setState('Waterkotte.Daten.Rohdaten', daten & 0xFF,true);
  port.close();
});
```

... tells me that you're closing the serial port as *soon* as you receive *any* data packet. You should give a bit of time for all the buffers to be flushed. Keep the port open until you're sure you got enough data.

(BTW, please indent your code, makes it easier to read)

wendy2702, 2018-02-28:
Hi Ivan,

thanks for answer.

I changed for testing to that:

```
port.on('data', function (data) {
	console.log('Data: ' + data);
	var daten=[];
	daten=data.toString('hex');
	setState('Waterkotte.Daten.Rohdaten', daten & 0xFF,true);
	//port.close();
```

Received this as feedback:

```

8:00:43.967 | [info] | javascript.0 script.js.Skript1: Connected
-- | -- | --
18:00:43.969 | [info] | javascript.0 script.js.Skript1: Daten gesendet
18:00:43.985 | [info] | javascript.0 script.js.Skript1: Data: �
18:00:44.003 | [info] | javascript.0 script.js.Skript1: Data: �)�)��
18:00:44.016 | [info] | javascript.0 script.js.Skript1: Data: �����R���v�A
18:00:44.032 | [info] | javascript.0 script.js.Skript1: Data: ��A�l�A��3B
18:00:44.049 | [info] | javascript.0 script.js.Skript1: Data: ll8BR8��pA
18:00:44.064 | [info] | javascript.0 script.js.Skript1: Data: '�@���?�5@:
18:00:44.079 | [info] | javascript.0 script.js.Skript1: Data: �A���@
18:00:44.095 | [info] | javascript.0 script.js.Skript1: Data: d@���@�v�A�
18:00:44.111 | [info] | javascript.0 script.js.Skript1: Data: �A�A��All
18:00:44.128 | [info] | javascript.0 script.js.Skript1: Data: 8B9�]F
18:00:44.143 | [info] | javascript.0 script.js.Skript1: Data: ��F�E
18:00:44.159 | [info] | javascript.0 script.js.Skript1: Data: 
18:00:44.176 | [info] | javascript.0 script.js.Skript1: Data: �S	%*
18:00:44.191 | [info] | javascript.0 script.js.Skript1: Data:  � N[
18:00:44.207 | [info] | javascript.0 script.js.Skript1: Data: ���C@��@�Q�
18:00:44.230 | [info] | javascript.0 script.js.Skript1: Data: >	oB��B�wA
18:00:44.243 | [info] | javascript.0 script.js.Skript1: Data: ??GB�A
18:00:44.256 | [info] | javascript.0 script.js.Skript1: Data: +�
18:00:44.273 | [info] | javascript.0 script.js.Skript1: Data: +5
18:00:44.288 | [info] | javascript.0 script.js.Skript1: Data: �@(
18:00:44.304 | [info] | javascript.0 script.js.Skript1: Data: ''(5
18:00:44.321 | [info] | javascript.0 script.js.Skript1: Data: ���

```

I was expecting the same HEX Code like it is displayed with Sniffer.

What else needs to be changed?

Thanks for support.

IvanSanchez, 2018-02-28:
Replace

```
	console.log('Data: ' + data);
```

with 

```
	console.log('Data: ', data);
```

...that will allow you to see things more clearly. if you use the `+` operator, you'll be casting everything into strings.

wendy2702, 2018-02-28:
After changing the above mentioned line it looks like this:

```

19:01:22.577 | [info] | javascript.0 script.js.Skript1: Daten gesendet
-- | -- | --
19:01:22.602 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.618 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.634 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.650 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.666 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.684 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.699 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.714 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.730 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.746 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.761 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.777 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.793 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.808 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.824 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.841 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.856 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.872 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.889 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.904 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.920 | [info] | javascript.0 script.js.Skript1: Data:
19:01:22.937 | [info] | javascript.0 script.js.Skript1: Data:

```

wendy2702, 2018-02-28:
If i change the logging to:

`console.log('Data: ' + data.toString('hex'));`

I got this:

```

19:35:56.809 | [info] | javascript.0 script.js.Skript1: Daten gesendet
-- | -- | --
19:35:56.836 | [info] | javascript.0 script.js.Skript1: Data: ff1002
19:35:56.852 | [info] | javascript.0 script.js.Skript1: Data: 00171fbe120c06030529a00dcc
19:35:56.868 | [info] | javascript.0 script.js.Skript1: Data: c0c36bd0c0fd73c0c0d525d7
19:35:56.883 | [info] | javascript.0 script.js.Skript1: Data: 4118dee341276e0b42d3ff33
19:35:56.899 | [info] | javascript.0 script.js.Skript1: Data: 42882b2e425238dec2000070
19:35:56.916 | [info] | javascript.0 script.js.Skript1: Data: 41007c9c3bfe348fc05ad313c1
19:35:56.931 | [info] | javascript.0 script.js.Skript1: Data: 70761442333373c0000000
19:35:56.947 | [info] | javascript.0 script.js.Skript1: Data: 00949c873f3f190641d525d741
19:35:56.963 | [info] | javascript.0 script.js.Skript1: Data: 18dee3410000904118dee34188
19:35:56.980 | [info] | javascript.0 script.js.Skript1: Data: 2b2e421320141c0212b4185d
19:35:57.008 | [info] | javascript.0 script.js.Skript1: Data: 4698dc1b461c018345000000
19:35:57.013 | [info] | javascript.0 script.js.Skript1: Data: 000101010101010101010101
19:35:57.028 | [info] | javascript.0 script.js.Skript1: Data: 010000000000c7530000020925
19:35:57.043 | [info] | javascript.0 script.js.Skript1: Data: 2a1212010d01e32000004e
19:35:57.061 | [info] | javascript.0 script.js.Skript1: Data: 5b1dc082e9434092e61740e151
19:35:57.076 | [info] | javascript.0 script.js.Skript1: Data: fe3e096f1442b58b0042cc771b
19:35:57.091 | [info] | javascript.0 script.js.Skript1: Data: 413f3f474202000000a041
19:35:57.107 | [info] | javascript.0 script.js.Skript1: Data: 00000000000000002cbe020002
19:35:57.123 | [info] | javascript.0 script.js.Skript1: Data: a32000200000000200000000
19:35:57.139 | [info] | javascript.0 script.js.Skript1: Data: 000000030108080000000000
19:35:57.155 | [info] | javascript.0 script.js.Skript1: Data: 000001270017012700283512
19:35:57.171 | [info] | javascript.0 script.js.Skript1: Data: 1003848aff

```

Is it somehow possible to get it as one big string or is it already on string and only splitted at log?

IvanSanchez, 2018-02-28:
It's up to you to concatenate all the data together. The usual thing to do is create a `Buffer` or an `Uint8Array` big enough, and put the incoming data there. When you have enough, do something with the data and empty the buffer.

What is the format of the data you're receiving?

wendy2702, 2018-02-28:
Hi,

data are coming as Hex Values. Output should be 243bytes.

```
Adresse	Anzahl	MenueEntry	MenueName_de
0x0000	0x0002	00.00	Versions-Nummer
0x0002	0x0003	00.01	Versions-Datum
0x0005	0x0003	00.02	Betriebs-Mode
0x0008	0x0004	01.00	Temp-Aussen
0x000C	0x0004	01.01	Temp-Aussen-24h
0x0010	0x0004	01.02	Temp-Aussen-1h
0x0014	0x0004	01.03	Temp-Ruecklauf Soll
0x0018	0x0004	01.04	Temp-Ruecklauf
0x001C	0x0004	01.05	Temp-Vorlauf
0x0020	0x0004	01.06	Temp-WW Soll
0x0024	0x0004	01.07	Temp-WW Ist
0x0028	0x0004	01.08	Temp-Raum
0x002C	0x0004	01.09	Temp-Raum-1h
0x0030	0x0004	01.10	Temp-WQuelle-Ein
0x0034	0x0004	01.11	Temp-WQuelle-Aus
0x0038	0x0004	01.12	Temp-Verdampfer
0x003C	0x0004	01.13	Temp-Kondensator
0x0040	0x0004	01.14	Temp-Saugleitung
0x0044	0x0004	01.15	Temp-frei
0x0048	0x0004	01.16	Druck-Verdampfer
0x004C	0x0004	01.17	Druck-Kondensator
0x00F3	0x0001	02.00	Handabschaltung
0x00F4	0x0004	02.01	Hzg Einsatzzeitpunkt Sollwert
0x00F8	0x0004	02.02	Hzg Ruecklauf Sollwert am Einsatzp.
0x00FC	0x0004	02.03	Kennline Steigung Soll
0x0100	0x0004	02.04	Kennline Obere Begrenzung
0x0050	0x0004	02.05	Hzg: Temp Ruecklauf Soll
0x0054	0x0004	02.06	Hzg: Temp Ruecklauf Ist
0x0104	0x0001	02.07	Sollwertanhebung 4k fuer 24 Stunden
0x0105	0x0004	02.08	Pilotraum Sollwert
0x0109	0x0001	02.09	Pilotraum Faktor
0x010A	0x0004	02.10	Externe Anhebung
0x010E	0x0003	02.11	Freigabe Heizung
0x0111	0x0003	02.12	Sperren Heizung
0x0114	0x0003	02.13	Sollwertaenderung Ein
0x0117	0x0003	02.14	Sollwertaenderung Aus
0x011A	0x0004	02.15	St. 2 Kl. Obere Begrenzung
0x011E	0x0004	02.16	Kennlinie Hysterese
0x0122	0x0001	02.17	Pumpen Nachlaufzeit
0x0123	0x0001	03.00	Abschaltung
0x0124	0x0004	03.01	Klg Einsatzzeitpunkt
0x0128	0x0004	03.02	Ruecklauftemp. Sollwert am Einsatzzeitp.
0x012C	0x0004	03.03	Klg Kennlinie Steigung Soll
0x0130	0x0004	03.04	Kennlinie untere Begrenzung
0x0058	0x0004	03.05	Klg: Temp Ruecklauf Soll
0x005C	0x0004	03.06	Klg: Temp Ruecklauf Ist
0x0134	0x0001	04.00	Abschaltung
0x0135	0x0003	04.01	Zeit ein
0x0138	0x0003	04.02	Zeit aus
0x0060	0x0004	04.03	WW: Temp Ist
0x013b	0x0004	04.04	Warmwasser Soll
0x013F	0x0004	04.05	Beckenwasser Temperatur Soll
0x0143	0x0004	04.06	Warmwasser Hysterese
0x0147	0x0004	04.07	Beckenwasser Hysterese
0x0064	0x0003	05.00	Uhrzeit
0x0067	0x0003	05.01	Datum
0x006A	0x0004	05.02	Betriebsstunden Kompressor
0x006E	0x0004	05.03	Betriebsstunden Heizbetrieb
0x0072	0x0004	05.04	Betriebsstunden Warmwasserbetrieb
0x0076	0x0004	05.05	Betriebsstunden Stufe 2
0x007A	0x0003	05.06	Messbeginn Zeit Kompressor
0x007D	0x0003	05.07	Messbeginn Datum Kompressor
0x0080	0x0003	05.08	Messbeginn Zeit Pumpen
0x0083	0x0003	05.09	Messbeginn Datum Pumpen
[... 62 lines cut ...]
0x00D1	0x0003	09.03	Gesamtstatus
0x00D4	0x0003	09.04	Status Verriegelung
0x00D7	0x0003	09.05	Status Heizung
0x00DF	0x0001	09.10	Mode Heizung
0x00E0	0x0001	09.11	Mode Kuehlung
0x00E1	0x0001	09.12	Mode Warmwasser
0x0000	0x0001	10.00	Logger Zyklus
0x0000	0x0001	10.01	Initialisierung des Loggers
0x0000	0x0004	10.02	Speicherfehler des Loggers
0x00E3	0x0003	10.03	Uhrzeit der Initialisierung
0x00E6	0x0003	10.04	Datum der Initialisierung
0x00E9	0x0002	10.05	Maximale Anzahl der Datensaetze
0x00EB	0x0003	10.06	Laufende Nr. aktueller Datensatz
0x00ED	0x0002	10.07	Anzahl gespeicherter Datensaetze
0x00EF	0x0002	10.08	Groesse eines Datensatzes
0x00F1	0x0002	10.09	Max Groesse das Datenspeicher
```

As example the `1fbe` from this row:

`19:35:56.852 | [info] | javascript.0 script.js.Skript1: Data: 00171fbe120c06030529a00dcc`

is the Version Number which i can check from Display an is 8126.

Do you need to know anything else?

wendy2702, 2018-03-01:
In the meantime i managed to put all data in one row using the Bytelength parser function but i noticed the the answer from Heating device varies from 252 to 256.

Is it somehow possible to make bytelength flexible?

And how can i resend my buffer to heating device each 60second without restarting complete script?

```
createState('Waterkotte.Daten.Rohdaten');

// To use the `ByteLength` parser:
const SerialPort = require('serialport');
const ByteLength = SerialPort.parsers.ByteLength;

var buffer = new Buffer(12);
buffer[0] = 0x10;
buffer[1] = 0x02;
buffer[2] = 0x01;
buffer[3] = 0x15;
buffer[4] = 0x00;
buffer[5] = 0x00;
buffer[6] = 0x00;
buffer[7] = 0xf3;
buffer[8] = 0x10;
buffer[9] = 0x03;
buffer[10] = 0x7c;
buffer[11] = 0x32;
const port = new SerialPort("/dev/ttyUSB0", {
    baudRate: 9600,
    dataBits : 8,
    parity : 'none',
    stopBits: 1
//  flowControl : false
}, function (err) {
    if (err) {
        return console.log('Error1: ', err.message);
    }

    const parser = port.pipe(new ByteLength({ length: 253 }));
    parser.on('data', function (data) {
		console.log('Data received: ' + data.toString('hex'));
		var daten = data.toString('hex');
		setState('Waterkotte.Daten.Rohdaten', daten, true);
		//port.close();
	});

	// Flush input
	port.flush();

	// send Data
	port.write(buffer, function(err) {
		if (err) {
			return console.log('Error on write: ', err.message);
		}
    
		console.log('Daten send');
// 		port.close();
    });
});

port.on('open', function() {
	console.log('Connected');
});

port.on('error', function(err) {
	console.log('Error2: ', err.message);
//	port.close();
});

// close port if the script stopped (to be able to open it again)
onStop(function (callback) {
	if (port && port.isOpen) {
		port.close();
		console.log('port closed');
	}
});

```

IvanSanchez, 2018-03-01:
> In the meantime i managed to put all data in one row using the Bytelength parser function but i noticed the the answer from Heating device varies from 252 to 256.

> Is it somehow possible to make bytelength flexible?

Parsing the data is up to you, unfortunately. You'll have to handle the buffers for incoming data manually.

If the protocol is based on request-response logic (i.e. the heater replies after you send it some commands), then you could for example implement some timers - output the entire response after X milliseconds without receiving any data.

> And how can i resend my buffer to heating device each 60second without restarting complete script?

See https://developer.mozilla.org/en-US/docs/Web/API/WindowOrWorkerGlobalScope/setInterval , and then

```
setInterval(function(){
    port.write(new Buffer([0x01, 0x02, 0x03]), function(err){
        // handle transmission errors
    });
}, 60000);
```

Please note that this sends data every 60 seconds. *This is different* than waiting 60 seconds after sending the previous piece of data, and is *different* than waiting 60 seconds after receiving data.

I recommend you do some research on how `setInterval()` and `setTimeout()` (along with `clearInterval()` and `clearTimeout()`) work.

wendy2702, 2018-03-01:
Hi Ivan,

thanks for answer. I'll try to find a solution for this:

``
If the protocol is based on request-response logic (i.e. the heater replies after you send it some commands), then you could for example implement some timers - output the entire response after X milliseconds without receiving any data.
``

Guess it will take some time till i understand how to do this.

wendy2702, 2018-03-07:
I was able to get the script working that it fits my needs. 

Thanks for support.

closed 2018-03-07 by wendy2702

reconbot, 2018-03-07:
Great! Want to share it?

wendy2702, 2018-03-07:
Please find below. Guess this can be coded in a better way but as I'm a beginner it's OK for me.

if you have any idea how to optimize the code please let me know.

```
createState('Waterkotte.Daten.Abfrage_Rohdaten');

// To use the `ByteLength` parser:
const SerialPort = require('serialport');
const ByteLength = SerialPort.parsers.ByteLength;

var buffer = new Buffer(12);
buffer[0] = 0x10;
buffer[1] = 0x02;
buffer[2] = 0x01;
buffer[3] = 0x15;
buffer[4] = 0x00;
buffer[5] = 0x00;
buffer[6] = 0x00;
buffer[7] = 0xf3;
buffer[8] = 0x10;
buffer[9] = 0x03;
buffer[10] = 0x7c;
buffer[11] = 0x32;
const port = new SerialPort("/dev/ttyUSB0", {
    baudRate: 9600,
    dataBits : 8,
    parity : 'none',
    stopBits: 1
//  flowControl : false
}, function (err) {
    if (err) {
//        port.close();
        return console.log('Error1: ', err.message);
    }

    const parser = port.pipe(new ByteLength({ length: 253 }));
    parser.on('data', function (data) {
		console.log('Data received: ' + data.toString('hex'));
		var daten = data.toString('hex');
		setState('Waterkotte.Daten.Abfrage_Rohdaten', daten, true);
		//port.close();
	});

	// Flush input
	port.flush();

	// send Data
	    port.write(buffer, function(err) {
		    if (err) {
			    return console.log('Error on write: ', err.message);
		    }
    
		    console.log('Daten send');
// 		    port.close();
        });
    });
    
port.on('open', function() {
	console.log('Connected');
});

port.on('error', function(err) {
	console.log('Error2: ', err.message);
//	port.close();
});

// close port if the script stopped (to be able to open it again)
onStop(function (callback) {
	if (port && port.isOpen) {
		port.close();
		console.log('port closed');
	}
});
```

reconbot, 2018-03-07:
I don't recommend flushing but the rest is pretty ok!

---
Francis Gulotta
wizard@roborooter.com

wendy2702, 2018-03-09:
Hi,

why not flushing the port?

Sometime the heating device response is longer then 253. Do you have any idea how to modify script to get complete message even it is longer?

Thanks

reconbot, 2018-03-09:
Flushing the port throws away data. Why flush the port? https://node-serialport.github.io/node-serialport/BaseBinding.html#flush

Data is received by the OS and given right away you can use parsers to emit it in chunks, or by patterns, etc. https://node-serialport.github.io/node-serialport/global.html#Parsers

wendy2702, 2018-03-09:
I tried with Regex and Delimiter Parser but didn't manage to get it working. My Problem is that the end of the received HEX Value allways looks like this:

``10 03 xx xx ff``

values for "xx xx" changes all the time.

I appreciate any kind of help.

reconbot, 2018-03-11:
This problem is harder than it looks.

Do you know the other kind of data it might send?

wendy2702, 2018-03-11:
Hi,

the data string allways starts with:

``
ff100200171fbe120c0
``

But the length and the content between start and end changes all the time. Im most cases the length is 253 but sometimes longer which depends on what the heating device currently is doing.

These are three examples with same length of 253:

```
ff100200171fbe120c06030529e69f7341d0244241b3dd6a41c596ae413120b241abaaba4100003442b1132d425238dec200007041ce5e5f4106566d412f3d9d4128b892410000a04100000000a0629440ebe68d40c596ae413120b241b45b9d413120b241b1132d4222330c0b0312c0715e46aef81c46a17b83450000000001010101010101010101010100000000005e9f00000a09252a1212010d01e32000004e5b1dc082e9434092e61740e151fe3e096f1442b58b0042cc771b413f3f474202000000a04100000000000000001b8103000000200005000068830000000000000000400808000000000000000127011501270028351210036543ff
ff100200171fbe120c060305295c0e7441d0244241b3dd6a41c596ae412176b1413a86ba4100003442f7fe2c425238dec200007041cd1860415afb6d416f089d41305e92410000a04100000000bf3d944074a98d40c596ae412176b141b45b9d412176b141f7fe2c4222340c0b0312c0715e46aef81c46a17b83450000000001010101010101010101010100000000005e9f00000a09252a1212010d01e32000004e5b1dc082e9434092e61740e151fe3e096f1442b58b0042cc771b413f3f474202000000a04100000000000000001b7c030000002000050000662b0000000000000000400808000000000000000127011501270028351210035931ff
ff100200171fbe120c060305298c3a7441d0244241b3dd6a41c596ae4138d8b0417d49ba41000034426ab62c425238dec200007041cbd26041038c6e412f3d9d410e1293410000a04100000000a062944063248e40c596ae4138d8b041b45b9d4138d8b0416ab62c4222350c0b0312c0715e46aef81c46a17b83450000000001010101010101010101010100000000005e9f00000a09252a1212010d01e32000004e5b1dc082e9434092e61740e151fe3e096f1442b58b0042cc771b413f3f474202000000a04100000000000000001b7a03000000200005000063d3000000000000000040080800000000000000012701150127002835121003fad9ff
```