serialport/node-serialport #1508: How do I restart .on('data') handler ?

scorpioRED, 2018-03-01:
- __SerialPort Version__: 6.1.1
- __NodeJS Version__: 8.6.0
- __Operating System__ and __Hardware Platform__: linux

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? 
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem

I try to create Class which should based on node-serialport.
This Class has methods which should send data through serial port then return answer from device (promise). As I understand, there is only one way to get answer from device (through serial port) it's use **serialport.on('data')** handler. But I need send end receive data many time. How do I do that correct , without direct answer form my request to device by **serialport .write** ?

### Steps and Code to Reproduce the Issue

`
class HardwareManager {
    constructor(path) {
        this.port = new SerialPort(path, {
            baudRate: 9600,
            dataBit: 8,
            autoOpen: false
        });

        this.port.open((err) => {
            if (err) {
                return console.log('Error opening port: ', err.message);
            }

        });

    }

    _runCMD(cmd) {
        let self = this;
        let counter = 0;
        let response = "";
        return new Promise((resolve, reject) => {
            self.port.write(cmd);
            self.port.on('data', (data) => {
                counter++;
                response = counter > 1 && counter < 4 ? response + data.toString('hex') : "";
                if (counter == 3) {
                    console.log(" stop answer for response");
                    self.port.write(Buffer.from("06", "hex"));

                    let toResolve = self._responseParser(response, cmdTitle);
                    counter = 0;
                    resolve(toResolve);
                }
    actionB(quontity) {
        let dataToSend = Buffer.from(this.formater(quontity) "hex")
        return this._runCMD(dataToSend, "deposit")
            })
        })
    }

    actionA() {
        let dataToSend = Buffer.from("045002", "hex");
        return this._runCMD(dataToSend)
    }

}

// and try to use it

let hardware = new HardwareManager("/dev/ttyS0");

hardware.actionA().then(answerA => {
    console.log("A answer is ->> ", answerA);
    hardware.actionB("1").then(answerB => {
        console.log("B answer is ->> ", answerB);
    })
})

`

It's work but I'm not sure is it correct, because I can't control port.on('data') handler, I'm afraid about memory leek after couple of thousands running methods actionA and ActionB

P.S.
My device should send me after request one response as
 <Buffer  01 50 02 45 30 33 30 33 31 31 30 30 03 15>

but I got 2 response  <Buffer 01 50 02 45 30 33 30 33> and <Buffer 31 31 30 30 03 15>
How can I fix it too ?

976032412, 2018-09-07:
@scorpioRED   我也有此问题，你解决了吗？

Hello-Tan, 2018-09-19:
I also had this problem.

closed 2018-11-25 by stale[bot]