serialport/node-serialport #1518: SerialPort makes app crash when a printer is plugged on the computer

AntoineFootbar, 2018-03-13:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__: 8.9.4
- __Operating System__ and __Hardware Platform__: Windows 7 Professional 64bits

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: Yes

### Summary of Problem
SerialPort makes app crash when a printer is plugged on the computer. When I plug my printer Brother MFC-J5620DW on my computer, my application which uses serialport for listing ports crashes instantly.

### Steps and Code to Reproduce the Issue
This is my function which list all my serial ports. When I comment it everything works great, but if I use it, when SerialPort.list() is called my app crashes. It crashes only if my printer (MFC-J5620DW) is plugged to my computer. This function is called when an USB device is connected to the computer. By crash I mean my electron becomes a blank page, devtools is deconnected and it does nothing. 
```javascript
/**
* List all trackers plugged and put it in a promise.
* @return {Promise.<Object, Error>} - list of trackers.
*/
export const getTrackers = () => {
  // Use serialport utility to list all serial ports.
  return SerialPort.list()
    .then(devices => {
      // Resolve the new list of devices, filtered by product and vendor id.
      return devices.filter(device => {
        return (
          device.productId && device.productId.includes('xxxx') &&
            device.vendorId && device.vendorId.includes('xxxx')
        );
      });
    });
};
```
Any idead about this issue ? Thanks !

xavier-breuil, 2018-03-23:
Update : I tried to reproduce this issue on different PCs, one with windows 10 and one with windows 7. On windows 10 everything works great but on windows 7 the bug is still here. I also tried to use your listing command "serialport-list", once again my devices are well listed on the windows 10 PC but not on the other one.

AntoineFootbar, 2018-03-23:
Update : I tried to reproduce this issue on different PCs, one with windows 10 and one with windows 7. On windows 10 everything works great but on windows 7 the bug is still here. I also tried to use your listing command "serialport-list", once again my devices are well listed on the windows 10 PC but not on the other one.

reconbot, 2018-04-28:
That's a driver thing with regards to the list. Without more output I can't help with debugging the crash. your approach looks valid. Sorry for the delay in response but
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2018-04-28 by reconbot

regedatimon, 2018-10-17:
I have exactly the same problem with Brother dcp8110dn on my electron app. Serialport version 4.0.7