serialport/node-serialport #1519: How to guarantee that all data receives will come in a single event

LorneZH, 2018-03-15:
- __SerialPort Version_4.0.7
- __NodeJS Version__9.7.0:
- __Operating System__ and __Hardware Platform__WIN10:

when used Parsers:
var SerialPort = require('serialport');
var port = new SerialPort('/dev/tty-usbserial1', {
  parser: SerialPort.parsers.raw
});

How to guarantee that all data receives will come in a single event.     (.on('data', callback))

reconbot, 2018-03-15:
I can't support v4 of SerialPort, it's old and buggy. I can tell you you'll
need a parser to decide how to breakup the data. Bytelength is usually a
good contender.

hanfengcan, 2018-03-15:
use \n at least in your data

LorneZH, 2018-03-16:
but I don't know the return data's length, and the POS method can't return \n

reconbot, 2018-03-16:
It's going to have some sort of protocol you'll need to parse.

reconbot, 2018-03-25:
There's usually a datasheet from the device manufacture that explains how the data is sent.

closed 2018-03-25 by reconbot