serialport/node-serialport #1520: term should be made into another package.

farfromrefug, 2018-03-16:
in the latest versions there are a lot of packages dependencies which are actually not needed when using node-serialport.
```commander```,  ```serialport-term``` and ```promirepl``` are there for the bin commands which you dont care about if you simply use serialport to develop your own device.

Those packages have a really big number of dependencies which makes apps bigger.
And removing deep dependencies is not easy.

Shouldn't it be better to make the bin commands into separated packages which depend on serialport?

reconbot, 2018-03-25:
I don't disagree, we recently added the menus but maybe we should pull them out.

reconbot, 2018-11-27:
done `@serialport/terminal`

closed 2018-11-27 by reconbot

farfromrefug, 2018-11-27:
@reconbot thanks!