serialport/node-serialport #1522: Installation of Serialport on Rock64/Ubuntu

Kunibert-007, 2018-03-17:
- __SerialPort Version__: 6 or 4 - tried with both of them
- __NodeJS Version__:  6.13.1
- __Operating System__ and __Hardware Platform__: Rock64 / Debian 9.4 // aarch64

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? yes. Python 2.7.13 is installed on the system.
- Are you using __Electron__ and have you checked the _Electron Docs_?: No Electron

### Summary of Problem

### Steps and Code to Reproduce the Issue
Install with npm install serialport result in the following error-log. Same result when I try it with the parameter --build-from-source. I'm not a linux-crack but for me it seems linke my linux can't find the installation-sources. But I don't know what I can do now. Please help! :-(
```
root@rock64:~# npm install serialport

> serialport@6.1.1 install /root/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.5.1
prebuild-install info looking for local prebuild @ prebuilds/serialport-v6.1.1-node-v48-linux-arm64.tar.gz
prebuild-install info looking for cached prebuild @ /root/.npm/_prebuilds/https-github.com-node-serialport-node-serialport-releases-download-v6.1.1-serialport-v6.1.1-node-v48-linux-arm64.tar.gz
prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/v6.1.1/serialport-v6.1.1-node-v48-linux-arm64.tar.gz
prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v6.1.1/serialport-v6.1.1-node-v48-linux-arm64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=6.13.1 runtime=node arch=arm64 platform=linux)
Traceback (most recent call last):
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py", line 13, in <module>
    import gyp
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 8, in <module>
    import gyp.input
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 5, in <module>
    from compiler.ast import Const
ImportError: No module named compiler.ast
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:336:16)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:219:12)
gyp ERR! System Linux 4.4.77-rockchip-ayufan-136
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/node_modules/serialport
gyp ERR! node -v v6.13.1
gyp ERR! node-gyp -v v3.6.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/root/package.json'
npm WARN root No description
npm WARN root No repository field.
npm WARN root No README data
npm WARN root No license field.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.1.1 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.1.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-03-17T11_01_34_946Z-debug.log

```

Kunibert-007, 2018-03-17:
When I try to install with the parameter `npm install serialport --build-from-source` I get the following log:
`root@rock64:~# npm install serialport --build-from-source

> serialport@6.1.1 install /root/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.5.1
prebuild-install info install --build-from-source specified, not attempting download.
Traceback (most recent call last):
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py", line 13, in <module>
    import gyp
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 8, in <module>
    import gyp.input
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 5, in <module>
    from compiler.ast import Const
ImportError: No module named compiler.ast
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:336:16)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:219:12)
gyp ERR! System Linux 4.4.77-rockchip-ayufan-136
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/node_modules/serialport
gyp ERR! node -v v6.13.1
gyp ERR! node-gyp -v v3.6.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/root/package.json'
npm WARN root No description
npm WARN root No repository field.
npm WARN root No README data
npm WARN root No license field.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.1.1 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.1.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-03-17T11_48_28_468Z-debug.log
`

Kunibert-007, 2018-03-17:
Ok, think I solved the issue myself. After executing `apt-get install python-dev` it seems to work. There were a few warnings but now serialport seems to work. Sorry for this issue, but maybe you can write the code for new install/reinstall of the python-dev-environment into the manual to help such linux-newbies like me :-)

fivdi, 2018-03-24:
Good to hear that it's working now.

The error message `"ImportError: No module named compiler.ast"` indicates that one prerequisites for [node-gyp](https://github.com/nodejs/node-gyp) was not installed. node-gyp is used to compile serialport when it's installed on ARM systems.

closed 2018-03-24 by fivdi