serialport/node-serialport #1523: My arduino uno doesn't receive data

banditFlex, 2018-03-17:
- __SerialPort Version__: v6.1.1
- __NodeJS Version__: v4.2.6
- __Operating System__ and __Hardware Platform__: Linux and Arduino

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? NO
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
I Everybody, here is my issue:
I'm using sails.js to start a nodejs serveur. I connect my arduino via usb, and connecting pin 11 to a led (with a resistor).

At some point i lauch the code that is suppose to open the serial connection et send 0 or 1 through usb, the arduino depending on the result should light up the LED.

I've tryined first with a check of the data read, then only if any data is read to light up the LED, but it seem the arduino never received anyting via the USB. I do see the small led on pcb for Rx lighting up, but it is never catch in the code.

Thank you for your advice.
Ben
### Steps and Code to Reproduce the Issue
Nodejs part:
                        var SerialPort = require('serialport');
			var port = new SerialPort('/dev/ttyACM0', {
				baudRate: 9600
			});

			port.on("open", function(){
				console.log("port opened");
				port.write("1\n");
			});

			// Open errors will be emitted as an error event
			port.on('error', function(err) {
				console.log('Error: ', err.message);
			});

Arduino part:

int value = 0;

void setup()
{
  Serial.begin(9600);
  pinMode(11, OUTPUT);
}
void loop()
{
  if(Serial.available()){
    value = Serial.parseInt();
    if(value == 1)
      digitalWrite(11, HIGH);
    else
      digitalWrite(11, LOW);
  }
  delay(1000);
}

fivdi, 2018-03-17:
An Arduino UNO does an automatic reset when it gets a serial connection. This means that it's necessary to wait for a few seconds before before sending data to the Arduino.

Does it work if the following JavaScript:
```js
		port.on("open", function(){
			console.log("port opened");
			port.write("1\n");
		});
```
is changed to:
```js
    port.on("open", function(){
        console.log("port opened");
        setTimeout(function(){
            port.write("1\n");
        },5000);
    });
```

banditFlex, 2018-03-17:
Hi fivdi,

First thank you for taking the time to answer me.
I've just test your solution, and it does work for me right now.
Thank you very much i've been stuck on this all morning.

reconbot, 2018-03-17:
Sending a ready byte when it boots up and using the ready parser also works.

closed 2018-03-24 by fivdi