serialport/node-serialport #1526: "Open (GetCommState): Unknown error code 1" with usb to serial conversion cable

nonoerg, 2018-03-20:
- __SerialPort Version__: 6.1.1
- __NodeJS Version__: 8.9.4
- __Operating System__ and __Hardware Platform__: Windows 10 x64

- Have you checked the right __version__ of the __api docs?__:
Yes.
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
Had no trouble installing. Checked the special case docs.
- Are you using __Electron__ and have you checked the _Electron Docs_?:
Yes. Using electron-rebuild v1.7.3.

### Summary of Problem
I'm using a USB to Serial conversion cable (TRENDnet TU-S9) (which could possibly be the cause for these issues) to communicate with a label printer (a Zebra GK420t).

The problem occurs when I'm trying to open a connection. The error I'm getting from the error listener is "Open (GetCommState): Unknown error code 1". The same error occurs when I try to connect with serialport-term (typing in "serialport-term" and picking a port).

I looked at the source code in `serialport_win.cpp` on github. The function where the error gets thrown seems to be `void EIO_Open(uv_work_t* req)` found [here ](https://github.com/node-serialport/node-serialport/blob/master/src/serialport_win.cpp) on line 92. Error code 1 means `ERROR_INVALID_FUNCTION` [according to microsoft](https://msdn.microsoft.com/en-us/library/windows/desktop/ms681382(v=vs.85).aspx).

The label printer definitely works. It comes with software that has some testing utilities which work just fine.

### Steps and Code to Reproduce the Issue
Add the files below and execute `npm install` and then `npm start`. The error gets logged to console immediately on my machine.

Folder structure:
```
/serialport demo
    package.json
    index.js
    index.html
```

Contents for package.json:
```json
{
  "name": "serialport-demo",
  "version": "1.0.0",
  "main": "index.js",
  "scripts": {
    "start": "electron .",
    "install": "electron-rebuild"
  },
  "dependencies": {
    "serialport": "^6.1.1"
  },
  "devDependencies": {
    "electron": "^1.8.4",
    "electron-builder": "^19.56.2",
    "electron-rebuild": "^1.7.3"
  }
}
```

index.js:
```javascript
'use strict';

const { app, BrowserWindow, ipcMain } = require('electron');
const path = require('path');
const url = require('url');
const serialport = require('serialport');

var mainWindow;

function createMainWindow() {
  let mainWindowOptions = {
    width: 800,
    height: 600,
    show: false,
    center: true
  };

  mainWindow = new BrowserWindow(mainWindowOptions);
  //mainWindow.webContents.openDevTools();

  mainWindow.on('closed', () => {
    mainWindow = null;
  });

  mainWindow.once('ready-to-show', () => {
    mainWindow.show();

    // Specify port here
    var port = 'COM3';
    var connection = new serialport(port);
    connection.on('error', (err) => {
      // Error: Open (GetCommState): Unknown error code 1
      console.log(err);
    });
  });

  mainWindow.loadURL(url.format({
    pathname: path.join(__dirname, 'index.html'),
    protocol: 'file:',
    slashes: true
  }));
}

app.on('ready', () => {
  createMainWindow();
});

app.on('window-all-closed', () => {
  if (process.platform !== 'darwin') {
    app.quit();
  }
});

app.on('activate', () => {
  if (!mainWindow) {
    createMainWindow();
  }
});
```

index.html

```html
<!DOCTYPE html>
<html>
<head>
	<meta charset="utf-8">
	<title>Serialport demo</title>
</head>
<body>
	<p>Hello world</p>
</body>
</html>
```

SirUppyPancakes, 2018-03-26:
Have you tried PuTTY or some other serial terminal app?
If either of those work, it may rule out a problem with the driver, but from reading here:
https://groups.google.com/forum/#!topic/microsoft.public.win32.programmer.ole/PrIyYoutyeE
it seems like it could be a driver thing.
If it is a problem with the driver, I'd try reinstalling from here:
https://www.trendnet.com/support/supportdetail.asp?prod=265_TU-S9

Another thing that couldn't hurt would be to read up on the documentation for the label printer and making sure all of your connection settings are correct (`baudRate`, `dataBits`, `stopBits`, etc.). If the docs are vague, you could always use a serial port monitoring utility to monitor their software and intercept the settings it uses (using a monitoring utility like this or similar: https://freeserialanalyzer.com/)

It also sounds like it's possible that any given driver may not support `GetCommState()`, though it seems like a pretty standard function and the docs don't seem to say anything about any alternatives...

nonoerg, 2018-03-27:
@SirUppyPancakes It does seem to be a driver issue.

If I use the software that comes with the printer everything works, I can issue commands through their software and the printer executes them (prints a label), but not through node-serialport.

I experimented a little - I uninstalled the software and drivers that come with the printer. After that Windows did its thing and installed whatever drivers were available in their online driverstore. After doing that I could connect to the printer through node-serialport however writing anything to it had no effect whatsoever. No errors, it behaved as if the data was successfully written.

I looked on the manufacturer site and apparently the adapter itself has been discontinued and the latest driver supports up to Windows 7.

I'll get the newer adapter and try again.

closed 2018-09-15 by reconbot