serialport/node-serialport #1532: How to disable ECHO on /dev/ttyUSBo

khvalov, 2018-03-22:
Is any chance to disable hardware echoing on /dev/ttyUSB0 device ? 
trick like: 

> const port = new SerialPort('/dev/ttyUSB0',{bindingOptions:{echo:false}});
does not help as well as stty -F /dev/ttyUSB0 -echo

reconbot, 2018-03-22:
There is no software echo, it's probably your device doing the echo.

khvalov, 2018-03-22:
Most probably yes, but my question - how to disable it (echoing)

reconbot, 2018-03-22:
I don't know what device you're using.

khvalov, 2018-03-23:
Mar 19 21:40:07 raspberrypi kernel: [   16.536172] usbserial: USB Serial support registered for FTDI USB Serial Device
Mar 19 21:40:07 raspberrypi kernel: [   16.536518] ftdi_sio 1-1.4:1.0: FTDI USB Serial Device converter detected
Mar 19 21:40:07 raspberrypi kernel: [   16.536725] usb 1-1.4: Detected FT232RL
Mar 19 21:40:07 raspberrypi kernel: [   16.544243] usb 1-1.4: FTDI USB Serial Device converter now attached to ttyUSB0

reconbot, 2018-03-23:
The device on the other end of your FTDI USB connection is probably sending
the data. There is no echo from your computer, you can't disable it from
software. If you're receiving the same data your sending, then whatever
hardware is on the other end of the serialport is sending it back to you.

---
Francis Gulotta
wizard@roborooter.com

khvalov, 2018-03-23:
Just made several tests and figure out that issue comes from "logic" itself. The device itself do not produce any echo (confirmed with several test in regular terminal app), 
Issue comes from "data" event handler. Once data sends to serial port, it's emit "data" event as well. So seems it's can't be used for "listening" data comes from serial

reconbot, 2018-03-23:
That's incorrect. Can you share some example code?

khvalov, 2018-03-23:
Here is the code 
cat ./serial.js
```
var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyUSB0', { autoOpen: true });
var i=0;

port.open(function (err) {
  if (err) {
    return console.log('Error opening port: ', err.message);
  }

  // write errors will be emitted on the port since there is no callback to write
  port.write('main screen turn on');
});

port.on('data', function (data) {
  console.log('Data: ' + data);
});

var timerId = setInterval(function() {
  port.write('hello'+i);
  i++;
}, 2000);
```

and output 
`nodejs ./serial.js
Error opening port:  Port is opening
Data: hello0
Data: hello1
Data: hello2
Data: hello
Data: 3
Data: hello4
Data: hello5
Data: hello
Data: 6
Data: h
Data: ello7
Data: hello8
Data: hello9
^C
pi@raspberrypi:~ $
`

fivdi, 2018-03-24:
@khvalov if `autoOpen: true` is specified when creating a `new SerialPort` instance then it's not necessary to call `port.open`.

I tried the test program exactly as posted above with a serial cable and can't reproduce the problem. The serial cable was not connected to anything at the other end and there was no echo. If there was a bug in serialport I would have expected there to be a echo.

If the transmit on the serial cable is directly connected to receive on the serial cable to create a loopback then there is of course what looks like an echo.

> Just made several tests and figure out that issue comes from "logic" itself. The device itself do not produce any echo (confirmed with several test in regular terminal app),

When you used the terminal app and hit a key, for example the key 'A', did you see 'A' printed to the screen? If you saw 'A' printed on the screen then the device at the other end of the serial connection is echoing the data it receives.

khvalov, 2018-03-24:
@fivdi , thanks for following this up. 
The major differences in our cases is probably - i do have exact eqipement hooked via serial (Cisco ASA 5510). 
As I mentioned - I have done few tests: 
- Using Windows + Putty - no echo when communicating with Cisco
- Raspberry PI + Screen - no echo when communicating with Cisco
- Raspberry PI + cu - no echo when communicating with Cisco 
Please find screenshot as evidence
code used 
`var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyUSB0');

port.open(function (err) {
  if (err) {
    return console.log('Error opening port: ', err.message);
  }
 
});

setTimeout(function(){
 port.write('help\n\n');
}, 3000);

port.on('data', function (data) {
  process.stdout.write(data.toString('utf8'));
});`

so as you can see port.write ('help\n\n') were exists on stdout (echoed...) 

[image]
[image]

fivdi, 2018-03-24:
> The major differences in our cases is probably - i do have exact eqipement hooked via serial (Cisco ASA 5510)

This is not the case.

While using the `cu` utility in the screenshot above you were making a **FALSE** assumption. You think that `cu` is printing `"help"` to the screen when you type it on the keyboard but this is not the case. cu is sending `"help"` via the serial port to the Cisco ASA 5510 and the Cisco ASA 5510 is echoing it back.

To help understand, please try the following command and post a screenshot of what you see and explain what you see after you understand it.
```
cu --halfduplex -l /dev/ttyUSB0 -s 9600
```

khvalov, 2018-03-24:
here is output. Seems you were right 
[image]

fivdi, 2018-03-24:
`--halfduplex` tells `cu` to do a local echo. 

Maybe there's a way to tell the Cisco ASA 5510 not to echo. If not, it looks like your program will need to take the echo into account and ignore it if it's not needed. 

There's no TODO for serialport here so I'll go ahead and close the issue.

closed 2018-03-24 by fivdi