serialport/node-serialport #1534: Binding TypeError exception and segmentation fault.

quezadaminter, 2018-03-25:
- __SerialPort Version__: 6.1.1
- __NodeJS Version__: v6.13.1
- __Operating System__ and __Hardware Platform__: Linux Arm7 (Maxwelll Aramada370 CPU)

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes, Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: No, No

### Summary of Problem
Mock binding test works, other scripts in serialport/bin/ fail with exception.
### Steps and Code to Reproduce the Issue
After building and installing serialport I ran the mocking.js script to test the setup and it works well:

admin@computer:/usr/local/lib/node_modules/npm/node_modules/serialport/bin$ ./mocking.js
Port opened:     COM_ANYTHING
Write:           Complete!
Last write:      Lets write data!
Received:        Hi from my test!READY
Received:        Lets write data!

However, running the echo.js program out of the box gives:
admin@computer:/usr/local/lib/node_modules/npm/node_modules/serialport/bin$ ./echo.js -p /dev/ttyUSB0
/usr/local/lib/node_modules/npm/node_modules/serialport/lib/serialport.js:113
    throw new TypeError('"Bindings" is invalid pass it as `options.binding` or set it on `SerialPort.Binding`');
    ^

TypeError: "Bindings" is invalid pass it as `options.binding` or set it on `SerialPort.Binding`
    at new SerialPort (/usr/local/lib/node_modules/npm/node_modules/serialport/lib/serialport.js:113:11)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/serialport/bin/echo.js:34:14)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.runMain (module.js:611:10)
    at run (bootstrap_node.js:387:7)
    at startup (bootstrap_node.js:153:9)

I looked around the serialport/lib folder to try and track down why the binding might be undefined or null and I modified the echo.js program slightly to force it to use the mock binding to try and isolate the problem by adding these lines at the top:

const SerialPort = require('../test'); // from inside the serialport repo
const MockBinding = SerialPort.Binding;

MockBinding.createPort(args.port, { echo: false, record: false });

It works well again:

admin@computer:/usr/local/lib/node_modules/npm/node_modules/serialport/bin$ ./echo.js -p /dev/ttyUSB0
echo: Port open: /dev/ttyUSB0
echo: READY!

Then I removed the mock binding from the echo.js file and forced it to use the linux binding with this code:

const Binding = require('../lib/bindings/linux');
const parsers = require('../lib/parsers');
SerialPort.Binding = Binding;
SerialPort.parsers = parsers;
module.exports = SerialPort;

But then I get a segmentation fault:

admin@computer:/usr/local/lib/node_modules/npm/node_modules/serialport/bin$ ./echo.js -p /dev/ttyUSB0
Segmentation fault

I am not sure if I am not calling the binding correctly and if there is a problem with the build/Release/serialport.node file. It could be it didn't build correctly for me but I am not sure how to debug further, I am stuck. If you have any suggestions on this I would appreciate getting pointed in the right direction.

reconbot, 2018-03-25:
If you use serialport as an external npm package do you have the same issue?

quezadaminter, 2018-03-26:
@reconbot:

Sorry, I am new to node.js and its modes of operation. Would you mind explaining how I use serialport as an external npm package to verify the behavior?

reconbot, 2018-09-15:
you make a package.json and install it via npm `npm install serialport`

closed 2018-11-22 by stale[bot]