serialport/node-serialport #1536: Read opened serialport settings

enginebeat, 2018-03-26:
- __SerialPort Version__: 5.0
- __NodeJS Version__: 8.2.1
- __Operating System__ and __Hardware Platform__: Windows 10 Desktop

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: Using Nodejs only

### Summary of Problem
After opening the port and getting communications going meaning that my program opened the port and is comunicating fine, I would like to know the settings of the port I am using.

### Steps and Code to Reproduce the Issue
This is my SerialPortController module:
`function SerialPortController(portSettings, dataReceived){
  var myPort;
  const SerialPort = require('serialport');
  const DelimiterParser = SerialPort.parsers.Delimiter;

  /* Should intialise the port with the passed settings */
  function InitPort(){

  }
  myPort = new SerialPort(portSettings.COMPortValue, {
    baudRate: Number(portSettings.baudrateValue),
    stopBits: Number(portSettings.stopbitsValue),
    dataBits: Number(portSettings.databitsValue),
    parity: portSettings.parityValue
  });

  const delimiterParser = myPort.pipe(new DelimiterParser({delimiter: ')'}));
  /* Should send back to the controller that the port is open */
  myPort.on('open', ()=>{
    console.log("Serial Port is Open");
    sendData('R');
    console.log(getPortInfo());
  });

  /* Deals with the data received */
  delimiterParser.on('data', (data)=>{
    dataReceived(data);
  });

  /* Need to use this for errors */
  myPort.on('err', (err)=>{
    console.log(err.message);
  });

    /* writes data to the serial port. needs to report errors */
  function sendData(data){
      //myPort.write(data, onError); need to have a look at the error
      myPort.write(data);
  };

  /* to get some info from the port */
  function getPortInfo(){
    if(myPort.open){
      var portSettings = {
        COMName: myPort.path,
        baudRate: myPort.baudRate,
        stopBits: myPort.stopBits,
        dataBits: myPort.dataBits,
        parity: myPort.parity
      };
      return portSettings;
    } else{
      return null;
    } 
  };

  return {
    sendData: sendData,
    getPortInfo: getPortInfo
  }
}
module.exports = SerialPortController;
/******************* ~ SerialPortController **************/`

On my main Module After creating an instance of SeriaPortController and giving it the settings for the ports, the port is open and comunicating fine.

I then use the method getPortInfo() and I get the settings back. The path and baudrate are fine but stopBits, startBits and parity come up as undefined.

This is not a big problem for me, I can store what values I passed to the port and get my values that way but I am just curious if this is something to do with the Windows 10 platform or if I'm doing this the wrong way, or even if this expected.
Thank you.
          Luis

enginebeat, 2018-03-27:
Closing issue, as SirUppyPancakes pointed out if you get the port.settings variable all the info for the port is there.
When I've done var portSettings =  myPort.settings;
I got this back:

{ autoOpen: true,
  baudRate: 115200,
  dataBits: 8,
  hupcl: true,
  lock: true,
  parity: 'none',
  rtscts: false,
  stopBits: 1,
  xany: false,
  xoff: false,
  xon: false,
  highWaterMark: 65536 }

so there is no issue.
thank you SirUppyPancakes.

closed 2018-03-27 by enginebeat

reconbot, 2018-03-27:
Great!