serialport/node-serialport #1542: is SerialPort fitting Raspbian?

eshk12, 2018-04-04:
hey guys i have an Raspbian OS installed on my raspberry,
i would like to know if it is ok to use this OS or I need to switch it.

`PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

reconbot, 2018-04-04:
I'm not really sure but our docs have link to a page with more details. https://github.com/node-serialport/node-serialport#raspberry-pi-linux

I imagine if it's not more than a year or two old then you'll be ok. I'd give it a try.

eshk12, 2018-04-04:
if i install ubuntu server instand of raspbian it will work without problem?

reconbot, 2018-04-04:
Both should work.

---
Francis Gulotta
wizard@roborooter.com

piechade, 2018-04-05:
Everything is perfectly working on rasbian, you have issues?

reconbot, 2018-04-28:
We have those instructions in our docs too. You could also not use root. But that's up to you and your application.

closed 2018-04-28 by reconbot