serialport/node-serialport #1550: File not found

noob-ckx, 2018-04-16:
platform：windows 10 64-bit
node:v6.1.0
python:2.7.14
here is my prompt when installing：
`> serialport@6.1.1 install D:\MyOwnItem\test\node_modules\serialport

prebuild-install || node-gyp rebuild

test@1.0.0 D:\MyOwnItem\test
-- serialport@6.1.1 extraneous
here is my code:
var SerialPort = require("serialport"); var portName = 'COM3'; var serialPort = new SerialPort( "COM3", { baudRate: 9600, dataBits: 8, parity: 'none', stopBits: 1, flowControl: false }, function(err){ console.log(err) });
then I got this:
Error: Opening COM3: File not found
at Error (native)

noob-ckx, 2018-04-17:
**OK，I got it ,This program can run when I connect a hardware on the serial port，but When my computer is not connected to hardware，the program will show 'File not found '
is this a bug ?  
I thought the program will open an unopened serial port？Why is it related to hardware?**

hanfengcan, 2018-04-17:
if you not connected to hardware, your pc has a  com3 ?

closed 2018-04-17 by noob-ckx