serialport/node-serialport #1551: Communication issues and random characters on Raspberry Pi

hpgmiskin, 2018-04-17:
- __SerialPort Version__: v6.1.1
- __NodeJS Version__: v8.9.4
- __Operating System__ and __Hardware Platform__: Linux version 4.9.59 on Raspberry Pi 3

- Have you checked the right __version__ of the __api docs?__: 👍 
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? 👎 
- Are you using __Electron__ and have you checked the _Electron Docs_?: 👎 

### Summary of Problem

When communicating with a Teensy microcontroller from a Raspberry Pi the serial communication does not perform as expected. I have created a sketch that performs an initial handshake with the computer then occasionally prints data over serial. In addition, the sketch will echo any message received over serial prefixed with `RECEIVED:` to distinguish the messages. The node script then quite simply opens the serial port and logs any messages received before closing the serial port after a period of time and repeating from the beginning.

When running the same code on my Mac I get the expected output from the program however when running on a Raspberry Pi the Teensy receives random messages from the computer. These messages generally include the characters `PORTCFG` alongside repeating characters `AT` with the occasional `#` which often delimits the both.

#### Logs on Mac 👍 

```
=== Running test 0
    BEGIN
    Some data
    Some data
    Some data
    Some data
    Some data
    Some data
    Some data
```

#### Logs on Raspberry Pi 👎 

```
=== Running test 0
    RECEIVED:BEGIN
    RECEIVED:P
    Some data
    RECEIVED:O
    RECEIVED:R
    RECEIVED:T
    RECEIVED:C
    RECEIVED:F
    Some data
    RECEIVED:G
    RECEIVED:?
    RECEIVED:A
    RECEIVED:T
```

What is seemingly happening is that the Teensy is not resetting in the same way when the port is closed when running on the Raspberry Pi. In addition, there are some characters that could be sent by the operating system when initiating communication. This part I am very unsure about, has anyone seen `PORTCFG` before when communicating over serial?

### Steps and Code to Reproduce the Issue

I have created a repository called [node-serialport-issue](https://github.com/hackscience/node-serialport-issue) for lack of a better name. The README includes instructions for installing and running the code locally and additionally how to deploy to a Raspberry Pi using `resin.io` which is a free service (when only using a couple of devices) which makes publishing code to Raspberry Pi very easy.

Please let me know if there is information missing or it is not possible to run the code example. I understand this could be an issue with Raspberry Pi or perhaps `resin.io` but thought that posting here someone could provide some insight.

hpgmiskin, 2018-04-17:
I have simplified the example used recreate the error by removing the `BEGIN` handshake from the Teensy setup function. In addition, I have made the sketch file work on either Teensy or Arduino. These changes are pushed onto the master branch of the repository.

Having deployed the updated code I no longer receive the `PORTCFG` message however on a number of test cases there is no data received from the Teensy. With DEBUG logs enabled for the serialport library, I get the following output, where I should be getting `BEGIN` followed by `Some data`.

```
=== Running test 0
  serialport:main opening path: /dev/ttyACM0 +6s
  serialport:bindings open +6s
  serialport:poller Creating poller +6s
  serialport:main opened path: /dev/ttyACM0 +11ms
  serialport:main _read reading +1ms
  serialport:bindings read +11ms
  serialport:unixRead Starting read +6s
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +4ms
  serialport:poller error Error: bad file descriptor +161ms
  serialport:main binding.read error Error: bad file descriptor +164ms
  serialport:main disconnected Error: bad file descriptor +1ms
  serialport:main #close +0ms
  serialport:bindings close +166ms
  serialport:main _read queueing _read for after open +1ms
  serialport:poller Stopping poller +3ms
  serialport:main binding.close finished +6ms
```

This issue does not present itself when programming with an Arduino so I am going to post on the Teensy forums as could well be a Teensy specific issue. Any support is much appreciated.

hpgmiskin, 2018-04-20:
This issue is not associated with `node-serialport` and  is actually due to `ModemManager` attempting to use the Teensy as a modem and thus blocking communication on the serial line. I cannot claim that this resolution is my own as was from this [resin forum post](https://forums.resin.io/t/usb-serial-communication-errors-between-raspberry-pi-and-teensy/2940/6).

I will close this issue but not remove the repository I have created as this could turn into a resource for others potentially in the future.

closed 2018-04-20 by hpgmiskin

reconbot, 2018-04-20:
This is awesome, I'm glad you figured it out. And I'm glad you did it in
public.