serialport/node-serialport #1553: Couldn't get enough info for serial devices with `node-serialport`

LockonS, 2018-04-18:
- __SerialPort Version__: 6.1.1
- __NodeJS Version__:9.11.1
- __Operating System__ and __Hardware Platform__: Mac OS X

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
`node-serialport` is nice, but it couldn't display enough port info. In my case I need to use two arduino devices at the same time, Uno and Micro. But Actually I couldn't tell which is Uno for sure unless I opened Arduino IDE to checkout or use platformio command to list the port information. `node-serialport` could display `comName`,`manufacturer`,`serialNumber`,`pnpId`,`locationId`,`vendorId`,`productId`, but none of these was able to tell me which device is what. Looking forward to see a resolution or a work around for this problem.

- This was the info displayed by `node-serialport`

```
port[0] { comName: '/dev/tty.usbmodem14131',
  manufacturer: 'Arduino LLC',
  serialNumber: undefined,
  pnpId: undefined,
  locationId: '14130000',
  vendorId: '2341',
  productId: '8037' }
port[2] { comName: '/dev/tty.usbmodem14141',
  manufacturer: 'Arduino (www.arduino.cc)',
  serialNumber: '554313134383515101B0',
  pnpId: undefined,
  locationId: '14140000',
  vendorId: '2341',
  productId: '0043' }
```
- This was the info displayed by platformio command, though not as much as `node-serialport` but I could easily tell which device was Arduino Micro.
```
/dev/cu.usbmodem14131
---------------------
Hardware ID: USB VID:PID=2341:8037 LOCATION=20-1.3
Description: Arduino Micro

/dev/cu.usbmodem14141
---------------------
Hardware ID: USB VID:PID=2341:0043 SER=554313134383515101B0 LOCATION=20-1.4
Description: Generic CDC
```

reconbot, 2018-04-18:
This is all the info we get from osx on the SerialPorts themselves. We
currently don't interrogate USB information which is why the Arduino app
can get more.

LockonS, 2018-04-18:
@reconbot Well then, thanks a lot for your reply. 😝 Seems that I need to figure another solution as a work around.

closed 2018-04-18 by LockonS