serialport/node-serialport #1554: setEncoding('hex') problem

ottomila, 2018-04-18:
- __SerialPort Version__: v6.1.1
- __NodeJS Version__: v8.11.1
- __Operating System__ and __Hardware Platform__: Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
I have a problem when setting the encoding 'hex'. When I use the read function it reads just 4bits at a time. For example I send 0x11 and the function serialPort.read(1) gives me "1".
If I use read(2) it works as it is supposed to.
Tthe moment I use read(n) with n>2 it's going back to reading just 4bits at a time.
Am I misinterpreting something about the "size" argument?

size | number | <optional> | Specify how many bytes of data to return, if available

### Steps and Code to Reproduce the Issue
I declared the serialport like this :
const serialPort = new serialport('/dev/ttyUSB0', { baudRate: 230400 }, (err) => {
    if (err) {
        log("Error: " + err.message);
    }
}).setEncoding('hex');

Using RealTerm I'm sending hex values, 0x11 0x14 0x35 0x35 0x35 0x35
console.log(serialPort.read(1)); //  1
console.log(serialPort.read(1)); //  1
console.log(serialPort.read(2)); // 14
console.log(serialPort.read(4)); // 3535
console.log(serialPort.read(4)); // 3535

ottomila, 2018-04-19:
Anyway eveything would be much easier if read(1) would actually read one byte when encoding is set to 'hex'.

reconbot, 2018-04-22:
Hex encoding returns a string and `read()` for string works on characters. This is a nodejs stream api thing. Glad you figured it out!

closed 2018-04-22 by reconbot