serialport/node-serialport #1560: Getting Error at: serialport.parsers.readline

devshiva619, 2018-04-30:
- __SerialPort Version__: 6.2.0
- __NodeJS Version__:  8.11.1
- __Operating System__ and __Hardware Platform__:

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? 
- Are you using __Electron__ and have you checked the _Electron Docs_?: Not using Electron

### Summary of Problem
I am running this on raspberrypi 3.
I have websocket version=5.1.1. 

When I am trying to run my `indigo-parking.js` I am getting an error.
```
pi@raspberrypi:~/Desktop/parking $ node indigo-parking.js
/home/pi/Desktop/parking/indigo-parking.js:16
    parser: serialport.parsers.readline("\n")
                               ^

TypeError: serialport.parsers.readline is not a function
    at Object.<anonymous> (/home/pi/Desktop/parking/indigo-parking.js:16:32)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:188:16)
    at bootstrap_node.js:609:3

```
### Steps and Code to Reproduce the Issue
`node indigo-parking.js`

[Link to File indigo-parking.js](https://github.com/taifur20/parkingsystem/blob/master/raspberry_pi/indigo%20parking/indigo-parking.js)

reconbot, 2018-04-30:
The link doesn't work. This error does sound like its expecting an older version of serialport.

devshiva619, 2018-04-30:
@reconbot this link works. https://goo.gl/cYwvmD

reconbot, 2018-04-30:
Have a look at the docs https://node-serialport.github.io/node-serialport/global.html#Parsers your code is using them a very old way.

devshiva619, 2018-05-01:
@reconbot thankyou for your help, I was able to run it. 👍

closed 2018-05-01 by devshiva619