serialport/node-serialport #1565: Race condition when opening a USB-CDC-ACM right after enumeration

IvanSanchez, 2018-05-07:
- __SerialPort Version__: 6.2.0
- __NodeJS Version__: 8.11.1
- __Operating System__: Linux (Debian testing, kernel 4.16.0).
- __Hardware Platform__: Any USB-CDC-ACM device (reproducible at least on a nRF52840 development board, and an Arduino Leonardo)

### Summary of Problem

The use case is opening an instance of `serialport` right after a USB CDC ACM device is plugged in (or rebooted). In other words: sending data to a USB device as soon as the device is powered on.

If the `serialport` is created in a event handler for [`node-usb`](https://www.npmjs.com/package/usb)'s `attach` event, the following error shows up:

``` 
  serialport:main opening path: /dev/ttyACM0 +0ms
  serialport:bindings open +0ms
  serialport:main Binding #open had an error Error: Error: Permission denied, cannot open /dev/ttyACM0 +3ms
```

### Steps and Code to Reproduce the Issue

Grab a hold of any USB CDC ACM device on a Linux box; copy-paste the following code into a file somewhere, run `npm add serialport usb debug`, run the file, detach/reattach (or reset, or power-cycle) your USB CDC ACM device.

```js
var Usb = require('usb');
var SerialPort = require('serialport');
var Debug = require('debug');
Debug.enable('*');

Usb.on('attach', function(usbDevice) {
    // Check that the newly attached USB device is the one we want
    // In a real case, this would check more stuff, and grab the USB serial number
     if (usbDevice.deviceDescriptor.idVendor !== 0x1915) {
         return;
     }

    // In a real case, call serialport.list(), check for a serial number matching 
    // the USB device's serial number, and get the comName from there
    var comName = '/dev/ttyACM0';

    // This will fail:
    var port = new SerialPort(comName);

    // However, this will work:
/*
    setTimeout(function(){
        var port = new SerialPort(comName);
    }, 50);
*/
});
```

Some further observations:

- The time needed for the linux drivers/devnodes/whoknows to settle down depends on the setup. In my case, it takes 25msec when connected to the root USB hub, and 12.5 when connected to a external hub.
- Running `fs.access()` will tell that the devnode is, in fact, readable and writable. However, running `fs.access` in a loop (or a `setInterval(..., 5)`) will make this race condition happen even with larger waits.
- It is possible to hit a harder, weirder race condition where the serial port is opened correctly, but no data is sent/received over USB (as far as my WireShark can tell). I haven't been able to isolate the conditions for that variant.
- I haven't tested this on win nor on mac. The race condition might (or hopefully might not) show up in those OSs.

bencefr, 2018-05-23:
This issue occurs on Windows 10 as well even with a delay of 1 second:
```
serialport:main opening +955ms path: COM37
serialport:bindings open +2s
serialport:main Binding #open had an error +2ms Error: Opening COM37: File not found
```

gniezen, 2018-06-06:
I just managed to reproduce it on Windows 8 too, with an FTDI device. We've had more than 70 users experience this error over the last 30 days (out of 920 attempts), so I was pretty happy to finally be able to reproduce myself and then finding this open issue. Note that this issue occurs even on v5.0.0 which is what we currently have on production. I was hoping that upgrading to v6.2.0 would fix it, but no such luck, so my guess is that this is an upstream bug.

Thanks @bencefr, I'm going to re-use the approach in your referenced PR to try multiple times to open the serial port before giving up.

reconbot, 2018-09-15:
It's going to take some time for the serialport bridge drivers to load after the usb device has been attached. I'm not aware of any other events you could listen for unfortunately. On windows I do know that the serial devices are "children" of the usb device and maybe there are specific events regarding that. In either case I'd just poll with `list()`.

reconbot, 2018-09-15:
I'm tempted to mark this as a bug, but I think it's more of an OS bug.

closed 2018-09-15 by reconbot