serialport/node-serialport #1566: Node 8.11 compile time warnings

MidnightWonderer, 2018-05-19:
- __SerialPort Version__: 6.2.0
- __NodeJS Version__: 8.11.2
- __Operating System__ and __Hardware Platform__: Raspbian on Raspberry Pi

### Summary of Problem
Compile time now shows a lot of warnings about callbacks.
```
prebuild-install WARN install No prebuilt binaries found (target=8.11.2 runtime=node arch=arm platform=linux)
make: Entering directory '/home/pi/project/node_modules/firmata/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:95:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterUpdate(uv_work_t*)’:
../src/serialport.cpp:150:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterClose(uv_work_t*)’:
../src/serialport.cpp:188:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterFlush(uv_work_t*)’:
../src/serialport.cpp:231:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterSet(uv_work_t*)’:
../src/serialport.cpp:285:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterGet(uv_work_t*)’:
../src/serialport.cpp:336:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterDrain(uv_work_t*)’:
../src/serialport.cpp:377:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
[... 85 lines cut ...]
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterDrain(uv_work_t*)’:
../src/serialport.cpp:424:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/poller.o
../src/poller.cpp: In static member function ‘static void Poller::onData(uv_poll_t*, int, int)’:
../src/poller.cpp:69:29: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   obj->callback.Call(2, argv);
                             ^
In file included from ../src/poller.cpp:1:0:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
  CXX(target) Release/obj.target/serialport/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/project/node_modules/serialport/build'
```

kcw-grunt, 2018-07-12:
This is the same behavior I am seeing as well. I was running node v4.6 and serialport 6.2.1 and everything was working as expected.  I wanted to duplicate the same env for the production platforms (Raspberry Pis) and realized they were running node v8 so i upgraded my dev platform to node v8....then most everything stops... What does work is no errors on init of the port and open and close emits are happening.  But, my code that ran write messages are not now being responded to.

reconbot, 2018-07-21:
Times keep on changing, it does work with these warnings. We'll have to upgrade to the newer methods.

zbjornson, 2019-01-06:
This was fixed by #1742/#1765.

HipsterBrown, 2019-01-08:
`serialport@7.1.2` has been published with those fixes. Closing this issue for now.

closed 2019-01-08 by HipsterBrown