serialport/node-serialport #1567: Module version mismatch. Expected 50, got 57.

Carsonsmuts, 2018-05-22:
- __SerialPort Version__: 6.2.0
- __NodeJS Version__: v8.4.0
- __Operating System__ and __Hardware Platform__:MacOS

### Summary of Problem

Getting error : Module version mismatch. Expected 50, got 57.

I understand this is a rebuild issue, however, deleting all node_modules and reinstalling / rebuilding never seems to rectify the issue. Is there any way I can force serialport to use v57 instead of v50?

Please note that I am running the app pre_compiled for testing purposes. 

Thanks in advance

### Steps and Code to Reproduce the Issue

Carsonsmuts, 2018-05-22:
I am aware that there are solutions such as....

npm rebuild --runtime=electron --target=1.4.9 --disturl=https://atom.io/download/atom-shell --build-from-source

It works, but I am not sure this is a good fix to the issue. If anyone could chime in on this I would appreciate it.

gniezen, 2018-06-06:
if you're using Electron, make sure that you're use the same Node version as the Electron runtime you're targeting. For example, if you're using Electron v1.8.4, you should build using Node v8.2.1.

reconbot, 2018-06-06:
@Carsonsmuts that works

closed 2018-06-06 by reconbot