serialport/node-serialport #1570: Putty works but not SerialPort

alaouiali, 2018-05-30:
- __SerialPort Version__: 6.2.0
- __NodeJS Version__: 9.9.0
- __Operating System__ and __Hardware Platform__: Win7 - CP2102

- Have you checked the right __version__ of the __api docs?__: yes.
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? yes.
- Are you using __Electron__ and have you checked the _Electron Docs_?: Not using electron

### Summary of Problem
I can read data from the COM port on Putty (and other terminals).
I am also able to correctly open the COM port on Node.
I am not able to receive ANY data from the device !
(I don't intend to write to the device, I only perform read operations)

### Steps and Code to Reproduce the Issue

This is the debug trace.
`  
serialport:binding:auto-detect loading DarwinBinding +0ms
serialport:main opening path: /dev/tty.SLAB_USBtoUART +0ms
serialport:bindings open +0ms
serialport:main _read queueing _read for after open +7ms
serialport:poller Creating poller +0ms
serialport:main opened path: /dev/tty.SLAB_USBtoUART +7ms
[/dev/tty.SLAB_USBtoUART]: open
serialport:main _read reading +0ms
serialport:bindings read +15ms
serialport:unixRead Starting read +0ms
serialport:unixRead waiting for readable because of code: EAGAIN +0ms
serialport:poller Polling for "readable" +2ms
`
The code:
`
// content of index.js
const http = require('http')
var SerialPort = require('serialport')

const portName = '/dev/tty.SLAB_USBtoUART'

const port = new SerialPort(portName, {
    baudRate: 115200,
  }).on('open', () => {
    console.log(`[${portName}]: open`)
  }).on('data', data => {
    console.log(`[${portName}]: data: ${data}`)
  }).on('error', err => {
    console.log(`[${portName}]: ${err}`)
    process.exit()
  })
`

alaouiali, 2018-05-30:
adding styling to the code.

KSR-Yasuda, 2018-06-28:
Are you running the script on COM5?
If you run the script on other console and not snatching COM5 input, I suppose you'd get no data.

alaouiali, 2018-06-28:
Thanks for the reply. The code is misleading because I tested this on Windows and Mac. I copied the debug trace from the Mac where the port name was "/dev/tty.SLAB_USBtoUART" and copied the node code from Windows "COM5". 

But to answer your question, listening to COM5 on a terminal or Putty.. I can see my data. PySerial library on python also works. 

I need to mention that other USB serial devices (like Arduino..) work perfectly with SerialPort.
Here I'm using a CP2102 from SiLabs, and SerialPort seems to not be able to handle communications with this device.

KSR-Yasuda, 2018-06-29:
So it is your device (SiLabs CP2102) specific problem...?

I just thought that if you're running the script on SSH or such, not `COM5`
(It is where I've stuck. `node-serialport` must run on the target port itself).

alaouiali, 2018-06-29:
It's a "node-serialport + Silabs CP2102" specific problem. because each one of them works perfectly but when put together, I can't get them to work.

I'm very doubtful someone can help here. Unfortunately, I had to drop node-serialport for this project and I really don't have time to debug. maybe in the future..
I'm closing the issue.

closed 2018-06-29 by alaouiali