serialport/node-serialport #1574: Package does not install on Raspberry Pi 3 with Node 10

basvdijk, 2018-06-10:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: v10.4.0
- __Operating System__ and __Hardware Platform__: Raspberry Pi 3, Raspbian GNU/Linux 9 

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
The package does not compile when using `npm i node-serialport`

### Steps and Code to Reproduce the Issue
Run `npm i node-serialport`

Error:

```
pi@Aker:~/rfxtest $ npm i serialport
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex

> serialport@4.0.7 install /home/pi/rfxtest/node_modules/rfxcom/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/node',
node-pre-gyp verb cli   '/home/pi/rfxtest/node_modules/rfxcom/node_modules/serialport/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@10.4.0 | linux | arm
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/home/pi/rfxtest/node_modules/rfxcom/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@10.4.0 (node-v64 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@10.4.0 (node-v64 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Connection closed while downloading tarball file
node-pre-gyp verb command build [ 'rebuild' ]
make: Entering directory '/home/pi/rfxtest/node_modules/rfxcom/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
make: Entering directory '/home/pi/rfxtest/node_modules/rfxcom/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘int getIntFromObject(v8::Local<v8::Object>, std::__cxx11::string)’:
../src/serialport.cpp:90:52: error: no matching function for call to ‘v8::Value::ToInt32()’
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
In file included from /home/pi/.node-gyp/10.4.0/include/node/node.h:63:0,
                 from ../../../../nan/nan.h:51,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/home/pi/.node-gyp/10.4.0/include/node/v8.h:2402:43: note: candidate: v8::MaybeLocal<v8::Int32> v8::Value::ToInt32(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Int32> ToInt32(Local<Context> context) const;
                                           ^~~~~~~
/home/pi/.node-gyp/10.4.0/include/node/v8.h:2402:43: note:   candidate expects 1 argument, 0 provided
In file included from /home/pi/.node-gyp/10.4.0/include/node/v8.h:26:0,
                 from /home/pi/.node-gyp/10.4.0/include/node/node.h:63,
                 from ../../../../nan/nan.h:51,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/home/pi/.node-gyp/10.4.0/include/node/v8.h:2415:34: note: candidate: v8::Local<v8::Int32> v8::Value::ToInt32(v8::Isolate*) const
                     Local<Int32> ToInt32(Isolate* isolate) const);
[... 761 lines cut ...]
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Linux 4.14.34-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/rfxtest/node_modules/rfxcom/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/rfxtest/node_modules/rfxcom/node_modules/serialport/build/Release"
gyp ERR! cwd /home/pi/rfxtest/node_modules/rfxcom/node_modules/serialport
gyp ERR! node -v v10.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN rfxtest@1.0.0 No description
npm WARN rfxtest@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-06-10T20_56_49_754Z-debug.log
```

JordanLos, 2018-06-14:
I had a similar problem a few weeks ago. This is the steps that got it working. I'm not an expert here, so some of these may be superfluous. 

0. Fix Locales 
Set the pi's locale to en_ca or en_us (since it comes from the U.K. its locale is off)
	$ sudo dpkg-reconfigure locales
		-use space to select en_CA.UTF-8 and deselect en_GB.UTF-8)
	$ sudo dpkg-reconfigure keyboard-configuration
	$ sudo dpkg-reconfigure tzdata

Source:
	- https://irrational.net/2012/04/18/making-the-raspberry-pi-less-british/

2. Fiddle with Unix stuff to get serialport working
This is a mess - DONT use nvm. apt-get also seems behind. Prefer this process on the pi
	a. add "unable_uart=1" to the file "/boot/config.txt"
	a. install node here: https://github.com/nodesource/distributions#debinstall for node version and use the script
		- Uninstalling nodejs/npm is a bit of a hassle. use "hash -r" to fix the problem on npm not sym linking when installing nodejs
	c. ensure you have build essentials
		$sudo apt-get install -y build-essential
	d. Get stuff to build serialport natively
		$sudo npm install -g node-gyp
		$sudo npm install -g node-pre-gyp
	e. Install serialport
		$sudo npm install serialport --unsafe-perm --build-from-source
	f. find location of node with $whereis node_modules
		$ cd node_modules/serialport
		$ sudo node-gyp configure build
Source: 
	- https://timodenk.com/blog/install-node-serialport-on-raspberry-pi/ (for uart config stuff)
	- https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots (for general nodejs)
	- https://github.com/nodesource/distributions#debinstall (location of installing nodejs)
	- https://libraries.io/npm/serialport (instructions for installing serialport properly)

basvdijk, 2018-08-26:
@fivdi I've contacted the package maintainer to upgrade the package.json to a more recent version. Thanks!

closed 2018-08-26 by basvdijk

adamreisnz, 2018-12-31:
I've run into the same issue yesterday with Node 10.15.0 and serialport 7.1.1 on an ARM unit.

Are there pre-built binaries for this combination yet? If not, where can we see what combinations are currently supported? Thanks!

adamreisnz, 2018-12-31:
I refactored to use the @serialport modules directly and tried again, appears to have worked now.

JEVBR, 2019-02-25:
> @basvdijk serialport@4.0.7 was released in December 2016 and is fairly old. It supports Node.js v8 but doesn't support Node.js v10. For Node.js v10 support please upgrade to serialport@6.2.2.

True, i had the same issue, Node V10, Ubuntu 18.04. by using the command:

`npm i serialport@latest`

the latest version is installed with no issues